nunobrum/PyLTSpice #4: Can't load .raw binary file

gpereira--, 2018-10-17:
Hi Nuno,

First of all, I want to thank you for this awesome piece of code! It was exactly what I was looking for and hopefully lots of other people will be helped too with your code.

I tested your approach with a very simple example (almost the same in your blog) just to test the functionality of loading a .raw file. Everything works just fine when I load a .raw ASCII file, but when trying to load a .raw binary file, things got weird and I really don't know what to do, probably my fault. 

Here is the code snippet and the following error:

```python
import LTSpice_RawRead
import matplotlib.pyplot as plt

LT = LTSpice_RawRead.LTSpiceRawRead('D:/Gustavo/MEGA/PyLTSpice-master/teste.raw')
IR1 = LT.get_trace("V(vsw)")
x = LT.get_trace(0)  # Zero is always the X axis
steps = LT.get_steps()
for step in range(len(steps)):
    # print(steps[step])
    plt.plot(x.get_wave(step), IR1.get_wave(step), label=steps[step])

plt.legend() # order a legend.
plt.show()
```
 And that's the output error message:
```
Found Numpy. WIll be used for storing data
Binary file
Normal access
Traceback (most recent call last):
  File "D:\Gustavo\MEGA\PyLTSpice-master\test_script.py", line 4, in <module>
    LT = LTSpice_RawRead.LTSpiceRawRead('D:/Gustavo/MEGA/PyLTSpice-master/teste.raw')
  File "D:\Gustavo\MEGA\PyLTSpice-master\LTSpice_RawRead.py", line 331, in __init__
    var.set_pointB(point, value)
  File "D:\Gustavo\MEGA\PyLTSpice-master\LTSpice_RawRead.py", line 75, in set_pointB
    self.data[n] = unpack("d", value)[0]
struct.error: unpack requires a buffer of 8 bytes
```

If necessary, I can also upload my .raw file. Best regards and Thank you very much.

nunobrum, 2018-10-17:
Dear Gustavo,

You're welcome. I'm really happy that people use and find my tools useful
and I really appreciate your message.
It's really hard to tell without having a look into the raw file. What kind
of simulation are you doing?
TRANsient, AC, OP or NOISE?
Another thing, there were some troubles to adapt between versions IV and
XVII.
Which is the LTSpice version you are using? Indeed, it would help to have a
look into your raw file.

Kind regards,
Nuno

gpereira--, 2018-10-17:
> Dear Gustavo, You're welcome. I'm really happy that people use and find my tools useful and I really appreciate your message. It's really hard to tell without having a look into the raw file. What [...]

Thanks for your response! It is a TRANsient simulation using the 64-bit version of LTspice XVII on Windows 10. My raw file is attached in the .zip archive below. Thank you again for your fast response!

[teste.zip](https://github.com/nunobrum/PyLTSpice/files/2489573/teste.zip)

nunobrum, 2018-10-21:
Dear Gustavo,

I've been debugging your raw file, and it seems that there is something
particular in your raw file that the current version is not supporting.
The file is almost half of the size of what the tool is expecting and the
error we are seeing is the tool trying to unpack a value that doesn't exist.
It seems at first glance that the data is stored in 32 bits instead of the
usual 64. Linear Technology (now ADI) has been making updates to the
LTSpice quite frequently, and last year the data format was changed from 32
to 64, so, I'm wondering if they eventually changed their minds, or if
there is something new.
Since, I've started this tool without documentation and pretty much had to
reverse engineer what to do, at this point, I'm trying to find a solution
so I can continue to support all versions.
I'll continue debugging this and will let you know as soon as I have a
solution.  It maybe will take a few more days.

Best Regards,
Nuno

gpereira--, 2018-10-22:
> Hi there, finally got it working again. since you don't have any steps done in your simulation I would recommend you use the code below. import LTSpice_RawRead import matplotlib.pyplot as plt LT = LTSpice_RawRead.LTSpiceRawRead('teste.raw') x = LT.get_trace(0) # Zero is always the X axis for trace in LT.get_trace_names(): # print(steps[step]) plt.plot(x.get_wave(), LT.get_trace(trace).get_wave(), label=trace) plt.legend() # order a legend. plt.show() input("Press enter") This will produce an image as shown in the linked file. [image] I will commit the latest change to GitHub right away. Best Regards, Nuno

Hi @nunobrum !

Thank you very much for your reply, and of course, your time! It worked just fine with your update and I got the same image you uploaded. There is only a single glitch that I will try to show you.

~~~Python
import LTSpice_RawRead
import matplotlib.pyplot as plt

LT = LTSpice_RawRead.LTSpiceRawRead('teste.raw')

x = LT.get_trace(0)  # Zero is always the X axis
traces = ['Ix(u3:DRAININ)','V(vsw)']
for trace in traces:
    plt.plot(x.get_wave(), LT.get_trace(trace).get_wave(),label=trace)

plt.legend()
plt.show()
~~~
This code above generates the following plot:
[image]
As you can note, the "x" axis shows time samples that are negative for some reason, maybe something related with the encoding. As I noted that the "y" axis displayed the right values, I simply changed the code above to the code below. Now I plot with the absolute value of "x" axis.

~~~Python
import LTSpice_RawRead
import matplotlib.pyplot as plt
import numpy as np

LT = LTSpice_RawRead.LTSpiceRawRead('teste.raw')

x = LT.get_trace(0)  # Zero is always the X axis
traces = ['Ix(u3:DRAININ)','V(vsw)']
for trace in traces:
    plt.plot(np.abs(x.get_wave()), LT.get_trace(trace).get_wave(),label=trace)

plt.legend()
plt.show()
~~~
[image]

And it's done! It may not be the best solution or an elegant solution, but it worked for plotting. As far as I know, it doesn't brought any kind of problems so far. The main issue with the ".raw" file you already solved and it made possible to continue with my research! I am truly grateful for your help!

Best Regards,

Gustavo

nunobrum, 2018-10-22:
Hello Gustavo,

I'm happy that it worked well. In regards to the bug you mentioned, this is
really weird.
More so because it seems that the problem is only felt for time values
inferior to 2us and superior to 0.5us.
It's quite strange and I confirm that I see exactly the same behaviour.
I'll investigate this sometime later and let you know. In any case, it
seems more a bug on the library I'm using for decoding.

By the way, I'm a bit curious about your name. What is your nationality?
You happen to have a Portuguese name.

Best Regards,
Nuno

gpereira--, 2018-10-23:
Yes, it is quite strange and it appears to happen just with time vector and nothing else. But yeah, we have this simple workaround by now, it may be a problem with the decoding library for sure.

And yes, I have a Portuguese name. I am Brazilian, born in Florianópolis. I am Electrical Engineer and currently working on my Master's Thesis research. I saw your name and thought you were Portuguese, but regarding documentation on GitHub I decided to write to you in English, so it could be helpful to others! 

Abraços!

Best regards,
Gustavo

closed 2018-11-01 by gpereira--