nunobrum/PyLTSpice #18: random time values interpreted as negative values

danielphili, 2020-02-10:
Dear Nunobrum,

Thanks for taking the effort of having started and maintaining this project.
I have following problem:

Somehow, some of time values in my raw file are being interpreted as below zero (which does not make sense in this case). 
The error occurs having simulated a simple circuit without any ".step param" statement or similar using LTSpiceIV.
It would be very nice if you could try to reproduce the error:
1) simulate the schematic "Draft4.asc" with LTSpiceIV and copy the "Draft4.raw" file to your "tests" folder.
2) In your "test_rawread.py" file, replace
`files = [
    'tran.raw',
    'tran - step.raw',
    'ac.raw',
    'ac - step.raw',
]` 
with `files = ['Draft4.raw']` and execute "test_rawread.py".
3) If you encounter negative values in time, you have reproduced the error.

Why I think this is critical, is that I cannot be sure if the time values are maybe not the only ones interpreted falsely negative. It does not seem like this is the case, however, since the wrong negative interpretation seems to be kind of rare (only few values), they might just be hidden in the simulation data where many values are so small you could not recognize them being falsely negative.

Thanks a lot for considering my issue!
Best regards,

Daniel (danielphili)

nunobrum, 2020-02-11:
Dear Daniel,

This issue with the negative values is actually a bug from LTSpice itself and discovered some time ago by Lee Johnston one of the biggest contributers of the project. Copying his findings below.

"Regarding the negative time values, I now have a transient analysis that
reproduces the problem - see attached. The root cause is actually LTspice.
I confirmed this by reading the raw, binary bytes using a different
software stack, National Instruments LabVIEW, and coerced the 8 byte words
to double precision float64. Both python and LabVIEW produce the same set
of float64 values. I believe the root cause is LTspice's 2nd order
compression algorithm. Disabling this feature via the control panel appears
to always produce positive values. I will follow-up with Analog Devices,
which now owns LTspice, to confirm it. As for the work around, I recommend
disabling 2nd order compression rather than taking the absolute value of
_traces[0] because this first trace isn't always time."

Disabling the second order compression solves it. So, because of that it wasn't ever solved on the code.

I hope this helps on your case too. 
Another possibility would be that you'd make at the end an abs() of the time vector. We don't do this on the code because in AC simulations this value is complex.

Good luck and happy simulations,
Nuno

danielphili, 2020-02-11:
Dear Nuno,

Thank you so much for taking the time and answering so quickly. I was getting a "little" bit unsure of myself when seeing this 😃 But now that this issue actually real and I am not being dumb, I am feeling better ;)

I might assess the larger file size disabling the 2nd order compression in my case and consider introducing a condition which applies abs() in case of a transient analysis.

Thank you once more. I'll close this issue then, since it is not related to your project.

Best regards,

Daniel

closed 2020-02-11 by danielphili