nunobrum/PyLTSpice #21: What to do if SPICE Circuit does not behave well ito convergence?

danielphili, 2020-07-06:
Dear nunobrum,

This is more a question than an issue.
I have built a fairly complex SPICE circuit. With some certain parameter configurations, it will take ages to converge or won't converge at all. Is there a simple way to make PyLTSpice identify such problems (fixed max simulation duration) and abort the one single problematic simulation and carry on with the next, ideally saving a raw file containing the data that has been recorded up to the point of abortion?
Best regards,

danielphili

nunobrum, 2020-07-06:
Dear Daniel,

I think what you are proposing is a good idea. However, I do need to
investigate how can it be done.
I must see whether LTSpice supports this in command lines, or whether the
library I am using to launch the LTSpice processes supports killing jobs,
or whether I need to write some code around it.
Since I'm a bit busy with my day work at this moment, what I can offer in
the mean time as a quick solution. Is to try to use the commands "SAVEBIAS"
and "LOADBIAS" to accelerate the convergence problems. This has helped me a
couple of times.
You run a simulation once and use the save bias in a point of your
choosing, and then you use the Loadbias to accelerate the other simulations.
The file format is text based, so you can even edit some values manually.

I hope this helps,
Nuno

danielphili, 2020-07-07:
Dear Nuno,

Thanks for considering! Have success during the implementation if that's a feasible feature. Unfortunately I am not familiar with such programming tasks, otherwise I would have offered my help..
Thanks for your tip also, I will look into it!
Best regards,

Daniel

nunobrum, 2020-07-21:
The implementation is done. Please check it and let me know if it fits your needs.

closed 2020-07-21 by nunobrum

danielphili, 2021-09-23:
Hi I have looked into it, because now, finally, a publication will come out of the work that I have done..
Since I needed the timeout option sort of explicitly, I implemented it myself and just worked on a local copy of your repo.
I updated your package today and saw that the way I needed it to be, the timeout is not yet included - or rather, I have not understood how to use it the way I want to. I have repeated the changes I made initially to make it work. I am not yet really good with all the options of git (forking, merging, branches etc..). But I would like to share my code alteration. Would you like me to send you the manipulation of LTSpiceBatch.py and LTSpice_RawRead.py? With a diff tool you could look into what I have changed (somewhat 20 lines or so).

nunobrum, 2021-09-23:
Hi Daniel,

If you can send me directly the code I can better understand what you wanted me to implement. 

Then I can either try to guide you on the pull request (not really obvious, git can be a bit obscure sometimes) or I can integrate it myself.  
As you prefer. 

Best regards,

Nuno

danielphili, 2021-09-24:
Hi Nuno,

Thank you so much :) I will attach the code files here.
Best,

Daniel
[LTSpice_RawRead.py.txt](https://github.com/nunobrum/PyLTSpice/files/7224549/LTSpice_RawRead.py.txt)
[LTSpiceBatch.py.txt](https://github.com/nunobrum/PyLTSpice/files/7224550/LTSpiceBatch.py.txt)

nunobrum, 2021-09-26:
Hi Daniel,

you did a really good job implementing the timeout feature.
I've integrated your code directly into the repo and will make a special
reference to you in the commit.

Thanks a lot,
Nuno