nunobrum/PyLTSpice #22: README description does not suit V1.0 - update proposal

danielphili, 2020-07-08:
Dear Nuno,

For version 1.0, you decided to move away from including a sim_settings.lib/.inc file. However, the README file still contains this as a description. Moving the description to a dedicated article describing "earlier versions" may be a useful clearification.
Also, it might be interesting to shortly summarize what needs to be done to move compatibility to the newest version of PyLTSpice. My suggestion:
1.) Remove/comment out ".inc sim_settings.lib" in the LTSpice schematic file
2.) Substitute any "LTC.set_settings(...)" call (with LTC being an instance of LTSpiceBatch.LTCommander) by "LTC.add_instructions(...)"
3.) Keep in mind that the .raw filename has change by the suffix "_run" for later processing steps.
Best regards and many thanks for your great work,

Daniel

akaeba, 2020-07-08:
Hey Daniel, this would be a nice first task one github. Fork, create a branch on your fork, „my-awesome-readme“ and create a pull request. The readme is written in markdown 😉

nunobrum, 2020-07-09:
I've updated the README file and I have made several corrections on the code including the point 3).
Just point 1) I was not able to address. I didn't find references to the sim_settings.lib,

closed 2020-07-09 by nunobrum