nunobrum/PyLTSpice #23: Feature Proposal:

danielphili, 2020-07-08:


akaeba, 2020-07-08:
Hey Daniel, yes I was also thinking about such an feature. Especially for parameter searches can it be interesting. But then would it also helpful to get an yaml file or whatever to postpone a parameter search and continue later.

nunobrum, 2020-07-09:
Thanks @danielphili for sharing your idea. I agree that it would be an awesome feature to have.
It did occur to me last time I've worked on this tool, but, it ended on the "forgot about it" bin. Good that you mention it (**).
Before answering to you, I did a quick googling about it before answer, and I think it can be done.

Hi @akaeba, can you please give me a bit more detail on your idea of the YAML file.  I'm not sure I understand what you are proposing.

(**) Thanks both for your interest in LTSpice batch. I was not sure people were really using it. It's the first time someone pokes me about it.

akaeba, 2020-07-09:
When I run the last parameter searches with pyltspice I face the issue that I need to abort the sim, cause of pc restart and continue later. Then it would helpful to store the parameter setting where to go on. With yaml you can write easily dicts into a file. This avoids to run some sims twice. Another point can be a python file for iterators . A one/two dimensional loop is easily but what about three or four parameters.

nunobrum, 2020-07-10:
Dear Andreas,

when you say last, do you mean the last version of the LTSpiceBatch, or the
last simulation you do with the tool?
Causing a PC to reset is something that really upsets me. That is something
that shouldn't ever happen.
What is your windows version?
There is in sorts already a log of executed simulations in the logs that
are saved, and my idea was that any kind of retry mechanism would be
handled by the controlling script.
Since that the tool cannot determine whether a simulation being executed is
a retry or not, I think we need two different commands. One for first time
simulation, and a second for retry failed simulations. What I do think is
that this would have to be done at a higher level.
For example, a SimulationManager super class that which usage would be
optional and generate YAML or JSON you are suggesting.
What do you think?

Best Regards,
Nuno

nunobrum, 2020-07-16:
Dear @akaeba and @danielphili 
I have a preliminary version of the LTSpiceBatch that is multi-threaded. 
Since it somehow breaks compatibility, I want to be careful with this upgrade.
In a multi-threaded environment, the user has to supply a callback function and this needs to be documented. I no longer can assume that anyone can understand the logic.

I'm still wondering about a pass/fail structure that makes sense. I need to find a way to know whether it's a retry with different settings or whether it is a brand new simulation. This is not really obvious. 
One hypothesis is to give as a parameter the old simulation result as a command line and only use the simulation number.

A more clever idea would be to see what was changed in regards to the original netlist, and if it was something that was already on the disk, simulation is not repeated. But, this quickly can become really complicated to know what was what. And what if it was just a simulation setting that changed. There is no way to know.

I'm a bit out of ideas. If you have some, please share.

akaeba, 2020-07-21:
Hey Nuno,

a simulation manager would a great deal. A step before i will try to create an iteration manager in pseudo code it looks  like in that way:

```python
iter.add(myStart1, myStop1, myInc1)
iter.add(myStart2, myStop2, myInc2)
iter.add(myStart3, myStop3, myInc3)

while (iter.done() /= True):
  param = iter.next()
```

This code will create a three dimensional loop. It realayes hopefully the creation of multidimensional loops. In this context i will also start to introduce an unit test.

What do you think?

nunobrum, 2020-07-21:
Dear Andreas,

the three dimensional loop can already be done with the existing code, but,
it looks a bit different from what you are proposing.

Here is how I envisioned it.

for var1 in sweep(start1, stop1, inc1):  # linear sweep
    sim_manager.set_parameters(param1=var1)  # sweeping a parameter
    for var2 in sweep(start2, stop2, inc2):
        sim_manager.set_component_value('Rx', var2)  # sweeping the value
of resistor Rx
        for var3 in sweep_log(start_freq, stop_freq, mult):  # sweeps in
logarithmic fashion
            sim_manager.set_parameters(freq=var3)
            sim_manager.run()

The sweep and sweep_log are iterators that you can find in the
LTSpiceBatch.py that I've included for the purpose of
making n-dimensional loops. I was planning on adding a sweep_npoints, and
sweep_log_npoints, where the user specifies the number of points on the
sweep instead of the increment.

I could implement what you are suggesting. It requires a bit less lines of
code, but it is less pythonic in its approach.
This means... a bit away from the Python language precepts.

Does this fit your needs?

Best Regards,
Nuno

akaeba, 2020-07-22:
Yes it would fit. I started my thinking from the point of avoiding to write down three loops. Especially if you like to comment one loop while the design process out. Then you have to rearrange all loops. In my opinion would the best way that the ltspice batch set component values would also except an dict with name and value. But for the first time I would suggest to move the sweep and sweep-log fiction into a own python file called iterator. That can improve the architecture cause ltspicebatch would then control und setup the sims only. If anybody like to have parametric searches it can moved into custom code.

nunobrum, 2020-07-22:
Since yesterday, I've been thinking about your idea and I think it is worth to implement it.
If having a lot of parameters to sweep, it becomes quite a messy code to have a lot of inner loops. 
I agree your idea makes the code a lot cleaner. I'll implement it next weekend.
Regarding the location of the iterators, I agree, that all of that would deserve its own module, as they are useful not only for simulations.

akaeba, 2020-07-22:
I did some similar in c, if you move the sweep functions into a own python module I can fork it and patch the functionality in it.

nunobrum, 2020-07-23:
@akaeba I have moved the sweep functions as agreed to its own module.
Unfortunately I will only have time to implement other stuff later this month.

akaeba, 2020-07-23:
Perfect, then I will start to add some functionality into the iterators module.

closed 2020-08-30 by nunobrum