nunobrum/PyLTSpice #26: _get_param_line stops at first match

jakmarikoukai, 2020-08-27:
Hi,

It seems that in a netlist with nested params definitions, the _set_parameter_ may not be able to find the right line to modify because the __get_param_line_ will stop as soon as a string match is found regardless of on wich side of the equal sign the param is.

E.g. in the following netlist : 
```
I1 0 N007 SINE(0 {I1p} 1k 0 0 0)

.param I1p=I1rms*sqrt(2)
.param I1rms=100

```

BR,
Jak

nunobrum, 2020-08-27:
Hello!
It's true! The existing code doesn't foresees that possibility. The good news is that it looks quite easy to fix.
Thanks for pointing that out. I'll send you a note as soon as it is fixed.

nunobrum, 2020-08-27:
Hi @jakmarikoukai 

I have made a fix to this problem of nested parameters. I did some tests on my side, and it seems to be working well.
Can you confirm it solves your problem ? If nothing reported in the next days, I'll push the change to PyPi.

Cheers,
Nuno

nunobrum, 2020-08-30:
Version pushed to PyPi

closed 2020-08-30 by nunobrum

jakmarikoukai, 2020-08-31:
Sorry for the late response. I confirm that it solves the problem.

Thanks !
Jak