nunobrum/PyLTSpice #27: set_parameter does not handle floating numbers

jakmarikoukai, 2020-08-31:
Hello Nuno,

When I use floating numbers with _set_parameter_, the netlist ends up with extra decimal separtors (e.g. : ".param I1=0.5.1").
If I'm not mistaken, the regexp is looking for alphanumeric characters (\w) but does not include the point, which makes the construction of the new line in the netlist a bit awkward.

```
 import re

# subsitute for set_parameters inputs
param = "I1"
value = "0.5"
line = ".PARAM I1=1.0"

regx = re.compile(r"%s\s*=\s*(\w*)" % param, re.IGNORECASE)
#line = self.netlist[param_line]
m = regx.search(line)
start, stop = m.span()
#self.netlist[param_line] = line[:start] + "{}={}".format(param, value) + line[stop:]
print(line[:start] + "{}={}".format(param, value) + line[stop:])
```

`=> .PARAM I1=0.5.0`

At first sight, the easiest way to fix it would be to simply build a new param line from the inputs, e.g. : 
`self.netlist[param_line] = ".param " + "{} = {}".format(param, value)`

But maybe, there is a case that I don't know and which would benefit from an enhanced regexp...

Best Regards,
Jak

nunobrum, 2020-08-31:
Hi Jak,

thanks for spotting this issue. Indeed the issue can be quickly solved by slightly change regex.
´´´
import re
# subsitute for set_parameters inputs
.param = "I1"
value = "0.5"
line = ".PARAM I1=1.0E+3"

regx = re.compile(r"%s\s*=\s*([\w\.+-]*)" % param, re.IGNORECASE)
#line = self.netlist[param_line]
m = regx.search(line)
start, stop = m.span()
#self.netlist[param_line] = line[:start] + "{}={}".format(param, value) + line[stop:]
print(line[:start] + "{}={}".format(param, value) + line[stop:])
´´´
now produces
.PARAM I1=0.5
as was expected.

nunobrum, 2020-08-31:
Hi @jakmarikoukai 

I've quickly fixed the issue you have found. Let me know if you find something else.
Anyway I'll have to refactor the code as soon as I have more time to do it.

All the best,
Nuno

closed 2020-08-31 by nunobrum