nunobrum/PyLTSpice #29: Clean-up Binary Files from repo

akaeba, 2021-01-10:
Hello Nuno,

in my opinion contains the repo some dirs with binary files, they are not mandatory to the repo:
* [__pycache__](https://github.com/nunobrum/PyLTSpice/tree/master/__pycache__)
* [.spyproject](https://github.com/nunobrum/PyLTSpice/tree/master/.spyproject)

I would suggest to remove them from the repo and add also to the [.gitignore](https://github.com/nunobrum/PyLTSpice/blob/master/.gitignore)

nunobrum, 2021-01-11:
Hello Andreas,

thanks for pointing it out. 
Yes, I agree we should remove these files.

Best Regards,
Nuno

nunobrum, 2021-01-11:
I have deleted the .spyproject.
Regarding the pycache, there was already an entry for the __pycache__ in the .gitignore. 
I don't understand why it's not simply streamed out.

closed 2021-01-11 by nunobrum

akaeba, 2021-01-11:
If you like to remove __pycache__ you have in your local dir to remove gitignore temporary. Then tracks git all files, now remove the cache and push the change w/o the change in the gitignore and then restore the gitignore.