nunobrum/PyLTSpice #30: LTSpice_Batch under macOS

simongapp, 2021-03-10:
Hi Nuno, 
I am trying to get to work your beautiful toolchain under macOS Catalina (10.15.7) in Jupyter notebook (Python version: 3.8.5) with LTspice XVII for OS X, Build Mar  3 2021 Version:17.0.23.0
I replaced line 37 in `LTSpiceBatch.py` with the macOS equivalent which gets rid of the `FileNotFoundError` message:
```
# Windows based original
#LTspiceIV_exe = [r"C:\Program Files (x86)\LTC\LTspiceIV\scad3.exe"]
LTspiceIV_exe = [r"/Applications/LTspice.app/Contents/MacOS/LTspice"]
```

I then tried to run the LTSpice_Batch example with my file Draft1.asc, which contains a simple voltage divider for testing purposes:
```
from PyLTSpice.LTSpiceBatch import LTCommander
from shutil import copyfile

# get script absolute path
meAbsPath = os.path.dirname(os.path.realpath("Draft1.asc"))
# select spice model
LTC = LTCommander(meAbsPath + "/Draft1.asc")
```
which opens a new instance of LTSpice but does not open the file. It only displays the start screen where you can either choose to open a new file or a recent file etc. 
The output is:
```
Creating Netlist
```
After manually exiting LTSpice, the output generates
```
The Netlist was successfully created
```
The LTC object contains the correct path to the .net file (from `LTC.netlist_file`) but the netlist ist empty (from `LTC.netlist`)

Aborting the process results in the following output:
```
---------------------------------------------------------------------------
KeyboardInterrupt                         Traceback (most recent call last)
<ipython-input-67-1caf2e289a03> in <module>
      5 meAbsPath = os.path.dirname(os.path.realpath("Draft1.asc"))
      6 # select spice model
----> 7 LTC = LTCommander(meAbsPath + "/Draft1.asc")

~/opt/anaconda3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py in __init__(self, circuit_file)
    719         warn("Deprecated Class. Please use the new SimCommander class instead of LTCommander\n"
    720              "For more information consult. https://www.nunobrum.com/pyspicer.html", DeprecationWarning)
--> 721         SimCommander.__init__(self, circuit_file, 1)
    722 
    723 

~/opt/anaconda3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py in __init__(self, circuit_file, parallel_sims)
    270 
    271             print("Creating Netlist")
--> 272             retcode = run_function(cmd_netlist)
    273             if retcode == 0:
    274                 print("The Netlist was successfully created")

~/opt/anaconda3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py in run_function(command)
    107 
    108     def run_function(command):
--> 109         result = subprocess.run(command)
    110         return result.returncode
    111 else:

~/opt/anaconda3/lib/python3.8/subprocess.py in run(input, capture_output, timeout, check, *popenargs, **kwargs)
    489     with Popen(*popenargs, **kwargs) as process:
    490         try:
--> 491             stdout, stderr = process.communicate(input, timeout=timeout)
    492         except TimeoutExpired as exc:
    493             process.kill()

~/opt/anaconda3/lib/python3.8/subprocess.py in communicate(self, input, timeout)
   1014                 stderr = self.stderr.read()
   1015                 self.stderr.close()
-> 1016             self.wait()
   1017         else:
   1018             if timeout is not None:

~/opt/anaconda3/lib/python3.8/subprocess.py in wait(self, timeout)
   1077             endtime = _time() + timeout
   1078         try:
-> 1079             return self._wait(timeout=timeout)
   1080         except KeyboardInterrupt:
   1081             # https://bugs.python.org/issue25942

~/opt/anaconda3/lib/python3.8/subprocess.py in _wait(self, timeout)
   1802                         if self.returncode is not None:
   1803                             break  # Another thread waited.
-> 1804                         (pid, sts) = self._try_wait(0)
   1805                         # Check the pid and loop as waitpid has been known to
   1806                         # return 0 even without WNOHANG in odd situations.

~/opt/anaconda3/lib/python3.8/subprocess.py in _try_wait(self, wait_flags)
   1760             """All callers to this function MUST hold self._waitpid_lock."""
   1761             try:
-> 1762                 (pid, sts) = os.waitpid(self.pid, wait_flags)
   1763             except ChildProcessError:
   1764                 # This happens if SIGCLD is set to be ignored or waiting

KeyboardInterrupt: 
```

I tried to manually add the .net file to LTC.netlist and change the value of resistor R2 with `LTC.set_component_value('R2', '10k')`. The changed value is visible in `LTC.netlist`. Running `LTC.reset_netlist()` does not produce any errors but `raw, log = LTC.run()`does. There seems to be an error where the path to the file occurs twice and the file itself is addressed as `Draft1_1.net`.
```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-10-c18aa03a086d> in <module>
----> 1 raw, log = LTC.run()
      2 LTC.wait_completion()

~/opt/anaconda3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py in run(self, run_id)
    747             # Write the new settings
    748             run_netlist_file = "%s_%i.net" % (self.circuit_radic, self.runno)
--> 749             self.write_netlist(run_netlist_file)
    750             cmd_run = LTspice_exe + LTspice_arg.get('run') + [run_netlist_file]
    751             print("cmd_run: " +str(cmd_run))

~/opt/anaconda3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py in write_netlist(self, run_netlist_file)
    601             if not line.endswith(END_LINE_TERM):
    602                 self.netlist[i] = line + END_LINE_TERM
--> 603         f = open(run_netlist_file, 'w')
    604         print("Self netlist: " +str(self.netlist))
    605         f.writelines(self.netlist)

FileNotFoundError: [Errno 2] No such file or directory: 'path_to_file/path_to_file_again/Draft1_1.net'
```

At this point I am out of ideas how to solve this issue and would appreciate help.

nunobrum, 2021-03-10:
Hello Simon,

Unfortunately LTSpiceBatch doesn't work in MAC OSX

https://electronics.stackexchange.com/questions/294310/start-a-ltspice-simulation-using-code-in-macos 
(copied the post below for your convenience)

After reading the post, I think that transposing all the features that are presently implemented in LTSpiceBatch to MAC OSX is going to be a big task.
Starting with the fact that the LTSpice version doesn't generate a netlist.  This means that all the functions that are now implemented to update directly the netlist won't work.
This still could be circumvented by generating the netlist manually by using the View-Netlist as you are probably doing, and this is supposed to work. 

The error is being raised on a file write. So, I guess it is something related  with access rights of the process running in regards to the folder you are using. Can you check this?

I hope this helps you,
Nuno

simongapp, 2021-03-15:
Hello Nuno,
it seems that calling LTSpice with `/Applications/LTspice.app/Contents/MacOS/LTspice -netlist Draft1.asc`has no other effect than that LTSpice starts.
Calling LTSpice with `/Applications/LTspice.app/Contents/MacOS/LTspice -Run Draft1.asc` starts LTSpice, opens the specified file (note: the absolute path is required here, otherwise LTSpice tries to open the file from the default LTSpice folder) and runs the simulation generating a Draft1.log, Draft1.net, Draft1.op.raw and a Draft1.raw file.
So I suspect it's not an access rights problem, but rather a problem with starting LTSpice correctly. 

For my purposes, it is enough to start LTSpice this way, modify the Draft1.asc file (with some string modifications, in my case I had to deal with utf-16-le and latin1 encondings), and then use your functions to read and further process the simulation data.
This way I can call LTSpice with to generate the simulation result files. I unchecked "Automatically delete .raw and .log files" in "Settings - Operation" to keep the files after the LTSpice process is terminated it also seems to help to call pid.kill() after waiting for a second. It might be possible to use some modified LTCommander() functions after that but I did not try that yet. 

```
import subprocess
import os
import time

def run_lt_spice(path_to_file):
    # check if file exists
    if (not os.path.isfile(path_to_file)):
        print("File: " + str(path_to_file) + " does not exist")
        return
    else:
        # Open LTSpice with file and run simulation to generate simulation files
        pid = subprocess.Popen(['/Applications/LTspice.app/Contents/MacOS/LTspice',
                                '-Run',
                                path_to_file])
        # Let some time pass in order to generate the .raw file etc.
        time.sleep(1)
        # Killing the process prevents the .net file from being deleted
        pid.kill()
        return pid
```

nunobrum, 2021-03-16:
Hello Simon,

The LTSpice was thought to rewrite netlist files, not ASC files.
Therefore, if we can't make the LTSpice in Mac to execute Netlists, then
the whole concept will not work.
I've already considered the option of handling ASC files, which are BTW
also text files, but this would become too dependent on the LTSpice
implementation. IF ever they will consider updating the ASC file format,
all that work would be lost.

So far I haven't seen any proof that the LTSpice implementation for Mac can
accept command lines.
Until that happens, I don't see much chances of automating your Spice
Simulations on Mac.
Other options is to run the Windows version either on a bootcamp drive, a
Virtual Machine, or maybe a Wine version for Mac.
Alternatively, you can check other Spice simulators.

P.S.,Believe it or not, I was an avid Mac user a few years ago. But
situations like the one you're facing together with the fact that Windows
made a giant leap in the last years, I don't see myself investing on a Mac
so soon.

Good luck,
Nuno

simongapp, 2021-03-16:
Hello Nuno,
thank you very much for the clarification, I was not aware of that!
Cheers!
Simon

closed 2021-03-16 by simongapp