nunobrum/PyLTSpice #31: raw file name not indexed properly

LukeGary462, 2021-04-05:
I am using PyLTSpice to do some automated circuit optimization and ran into an issue with pulling the raw data from batch simulations. 

See the output below:
```
$ pipenv run test.py
Found Numpy. WIll be used for storing data
Creating Netlist
The Netlist was successfully created
Mon Apr  5 12:51:54 2021 : Starting simulation 1
Mon Apr  5 12:51:56 2021: Simulation Successful. Time elapsed 00:00:00:

D:\Documents\test-dir\test-ckt-optimization\circuits\test-ckt-rev-A1_run.raw
D:\Documents\test-dir\test-ckt-optimization\circuits\test-ckt-rev-A1_1.log
Traceback (most recent call last):
  File "D:\Documents\test-dir\test-ckt-optimization\test.py", line 46, in <module>
    result = LTSpiceRawRead(raw)
  File "C:\Users\luke\.virtualenvs\test-ckt-optimization-vibq49oT\lib\site-packages\PyLTSpice\LTSpice_RawRead.py", line 264, in __init__
    raw_file_size = os.stat(raw_filename).st_size  # Get the file size in order to know the data size
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'D:\\Documents\\test-dir\\test-ckt-optimization\\circuits\\test-ckt-rev-A1_run.raw'
```

I can remedy the issue by doing a string replace of the raw file to replace `_run.raw` with `_{index}.raw` and get the behavior I want. Looking into the source, its seems that the raw file output is not updated like the log file output name is in `LTCommander::run`
```python
    def run(self, run_id=None):
        """
        Executes a simulation run with the conditions set by the user. (See also set_parameter, set_component_value,
        add_instruction)
        The run_id parameter can be used to override the naming protocol of the log files.
        :return (raw filename, log filename) if simulation is successful else (None, log file name)
        """
        # update number of simulation
        self.runno += 1  # Using internal simulation number in case a run_id is not supplied

        # decide sim required
        if self.netlist is not None:
            # Write the new settings
            run_netlist_file = "%s_%i.net" % (self.circuit_radic, self.runno)
            self.write_netlist(run_netlist_file)
            cmd_run = LTspice_exe + LTspice_arg.get('run') + [run_netlist_file]

            # run the simulation
            start_time = clock_function()
            print(time.asctime(), ": Starting simulation %d" % self.runno)

            # start execution
            retcode = run_function(cmd_run)

            # process the logfile, user can rename it
            dest_log = "%s_%i.log" % (self.circuit_radic, self.runno)
            # print simulation time
            sim_time = time.strftime("%H:%M:%S", time.gmtime(clock_function() - start_time))
            # handle simstate
            if retcode == 0:
                # simulation succesfull
                print(time.asctime() + ": Simulation Successful. Time elapsed %s:%s" % (sim_time, END_LINE_TERM))
                self.write_log("%d%s" % (self.runno, END_LINE_TERM))
                self.okSim += 1
            else:
                # simulation failed
                self.failSim += 1
                # raise exception for try/except construct
                # SRC: https://stackoverflow.com/questions/2052390/manually-raising-throwing-an-exception-in-python
                # raise ValueError(time.asctime() + ': Simulation number ' + str(self.runno) + ' Failed !')
                print(time.asctime() + ": Simulation Failed. Time elapsed %s:%s" % (sim_time, END_LINE_TERM))
                # update failed parameters and counter
                dest_log += 'fail'

            try:
                os.replace(self.circuit_radic + '_run.log', dest_log)
            except FileNotFoundError:
                pass

            if retcode == 0:  # If simulation is successful
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
                return self.circuit_radic + '_run.raw', dest_log  # Return rawfile and logfile if simulation was OK
+++++++++++++++++++++++ CHANGE TO ++++++++++++++++++++++++++++++++++++++++++
                raw_name = self.circuit_radic + '_' + str(self.runno), dest_log
            else:
                return None, dest_log
        else:
            # no simulation required
            raise UserWarning('skipping simulation ' + str(self.runno))
```

after doing the file name string replacement I can get clean output
```
$ pipenv run test.py
Found Numpy. WIll be used for storing data
Creating Netlist
The Netlist was successfully created
Mon Apr  5 12:55:24 2021 : Starting simulation 1
Mon Apr  5 12:55:26 2021: Simulation Successful. Time elapsed 00:00:00:

D:\Documents\test-dir\test-ckt-optimization\circuits\test-ckt-rev-A1_run.raw
D:\Documents\test-dir\test-ckt-optimization\circuits\test-ckt-rev-A1_1.log
Normal access
['frequency', 'gain', 'V(d1.rs)', 'V(d1.sid)', 'V(d1.fid)', 'V(d1)', 'V(r9)', 'V(r3)', 'V(r12)', 'V(r11)', 'V(r8)', 'V(r7)', 'V(r6)', 'V(r5)', 'V(r4)', 'V(r2)', 'V(r1)', 'V(onoise)', 'inoise']
```

Just thought I would notify of the bug here, or perhaps I am misunderstanding the module a tad.

nunobrum, 2021-04-06:
Hi Luke,

I think you're right. The way the netlist name is being created is not
compatible with the rest of the code.
I should note however, that you are using a class that is deprecated, which
is maintained for backward compatibility. For that reason, the error was
not trapped before.

I think that it can be quickly fixed by changing the line 383 from

run_netlist_file = "%s_%i.net" % (self.circuit_radic, self.runno)

to

run_netlist_file = "%s.net" % (self.circuit_radic)

I haven't checked it because I'm on the go.
I'll check and commit as soon as I can.

Cheers,
Nuno

LukeGary462, 2021-04-06:
Great. With regards to the deprecated class, will the documentation ever be updated to be a little more complete?

nunobrum, 2021-04-06:
☺️ sorry about that. You’re right. 
All that deserves a far better documentation. I’ll add it high ranking in my to do list. 

Nuno

LukeGary462, 2021-04-09:
This is a great tool! I would love to see more support and help in developing this further

nunobrum, 2021-06-07:
Hi Luke,

I've just committed a more comprehensive documentation to github.
Still need to connect it to readthedocs, if you have sphinx you can already
generate the documentation you need.

Best Regards,
Nuno

closed 2021-06-14 by nunobrum