nunobrum/PyLTSpice #32: Regression in set_component_value function when line contains a parameter substitution

travjp, 2021-06-29:
Steps to reproduce:
```
(venv) C:\work>pip install git+https://github.com/nunobrum/PyLTSpice.git#egg=PyLTSpice
...
(venv) C:\work>python
Python 3.6.2 (v3.6.2:5fd33b5, Jul  8 2017, 04:57:36) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyLTSpice.LTSpiceBatch import SimCommander
>>> LTC = SimCommander("my_circuit.asc")
Creating Netlist
The Netlist was successfully created
>>> LTC.add_instruction("R1 A B {Rparam}")
>>> LTC.set_component_value("R1", 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\work\venv\lib\site-packages\PyLTSpice\SpiceEditor.py", line 383, in set_component_value
    self._set_model_and_value(device, value)
  File "C:\work\venv\lib\site-packages\PyLTSpice\SpiceEditor.py", line 262, in _set_model_and_value
    raise NotImplementedError('Unsupported line "{}"\nExpected format is "{}"'.format(line, regxstr))
NotImplementedError: Unsupported line "R1 A B {Rparam}
"
Expected format is "^(R\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(3).*}|([0-9\.E+-]+(Meg|[kmunp])?R?))).*$"
>>>
```

Expected behavior:
```
(venv) C:\work>pip install PyLTSpice==1.1
...
(venv) C:\work>python
Python 3.6.2 (v3.6.2:5fd33b5, Jul  8 2017, 04:57:36) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyLTSpice.LTSpiceBatch import SimCommander
>>> LTC = SimCommander("my_circuit.asc")
Creating Netlist
The Netlist was successfully created
>>> LTC.add_instruction("R1 A B {Rparam}")
>>> LTC.set_component_value("R1", 1)
>>>
```

It appears that in 56c9f65, the group number wasn't updated in the conditional for the 'R','L','C' in REPLACE_REGXES in PyLTSpice/SpiceEditor.py.

nunobrum, 2021-07-02:
Thanks for your message. Indeed your case is not being covered, and I'm not sure why the previous code didn't give an error, as the  previous regex was not covering that situation either.  For reason I need to investigate, it wasn't raising an error.
I'll make the correction, but may I just ask why don't you just update the parameter value instead of the resistor value.
I mean, you already have the Rparam already defined, so you could do a set_parameter('Rparam', <value>)

travjp, 2021-07-02:
Looks like I was running the previous version 1.1 (Aug 30, 2020) from PyPI which was released right before 56c9f65. This commit added a group to the element name match, e.g. ^(R\w+) therefore changing the group number for the curly bracket match.
In my circuit, I have many components with calculated values in which I'm looping through and selectively overriding each. Giving each component a param and updating that instead would definitely work for me as well. As far the Rparam example, it was the simplest test case I could come up with. Thanks for your work on this package; it's much appreciated!

```
From: travjp <>
Date: Tue, 29 Jun 2021 13:59:53 -0500
Subject: [PATCH] Update referenced group in conditional to the group matching
 the opening curly bracket in the RLC replace regexes. Fixes regression from
 previous behavior.

---
 PyLTSpice/SpiceEditor.py | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/PyLTSpice/SpiceEditor.py b/PyLTSpice/SpiceEditor.py
index 24246fc..f4e321b 100644
--- a/PyLTSpice/SpiceEditor.py
+++ b/PyLTSpice/SpiceEditor.py
@@ -27,15 +27,15 @@ UNIQUE_SIMULATION_DOT_INSTRUCTIONS = ('.AC', '.DC', '.TRAN', 'NOISE', '.DC', '.T
 
 REPLACE_REGXES = {
     'B': r"^(B[VI]\w+)(\s+[\w\+\-]+){2}\s+(?P<value>.*)$",  # Behavioral source
-    'C': r"^(C\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(3).*}|([0-9\.E+-]+(Meg|[kmunp])?F?))).*$",  # Capacitor
+    'C': r"^(C\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(4).*}|([0-9\.E+-]+(Meg|[kmunp])?F?))).*$",  # Capacitor
     'D': r"^(D\w+)(\s+[\w\+\-]+){2}\s+(?P<value>\w+).*$",  # Diode
     'I': r"^(I\w+)(\s+[\w\+\-]+){2}\s+(?P<value>.*)$",  # Current Source
     'J': r"^(J\w+)(\s+[\w\+\-]+){3}\s+(?P<value>\w+).*$",  # JFET
     'K': r"^(K\w+)(\s+[\w\+\-]+){2:4}\s+(?P<value>[\+\-]?[0-9\.E+-]+[kmunp]?).*$",  # Mutual Inductance
-    'L': r"^(L\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(3).*}|([0-9\.E+-]+(Meg|[kmunp])?H?))).*$",  # Inductance
+    'L': r"^(L\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(4).*}|([0-9\.E+-]+(Meg|[kmunp])?H?))).*$",  # Inductance
     'M': r"^(M\w+)(\s+[\w\+\-]+){3}\s+(?P<value>\w+).*$",  # MOSFET
     'Q': r"^(Q\w+)(\s+[\w\+\-]+){3}\s+(?P<value>\w+).*$",  # Bipolar
-    'R': r"^(R\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(3).*}|([0-9\.E+-]+(Meg|[kmunp])?R?))).*$",  # Resistors
+    'R': r"^(R\w+)(\s+[\w\+\-]+){2}\s+(?P<value>({)?(?(4).*}|([0-9\.E+-]+(Meg|[kmunp])?R?))).*$",  # Resistors
     'V': r"^(V\w+)(\s+[\w\+\-]+){2}\s+(?P<value>.*)$",  # Voltage Source
     'X': r"^(X\w+)(\s+[\w\+\-]+){1,99}\s+(?P<value>\w+)(\s+\w+\s*=\s*\S+)*$",  # Sub-circuit
 }
-- 
2.17.1
```

nunobrum, 2021-07-03:
Thanks for the correction. Well done. It's really not obvious to debug regex.
Are you going to push it to the repo ? 
If not, I have already updated locally as per your corrections and I commit it myself.

closed 2021-07-03 by nunobrum