nunobrum/PyLTSpice #35: Running under Linux

bernienor, 2021-09-25:
I have problems running this library under Linux. 

I have installed LTspice using wine. And pulled the sources for PyLTspice from github. I changed the definition of LTspice_exe to 'wine ~/.wine/Program Files/LTC... but I can't find a way to get this to run due to the space in the path to LTspice. I could move the location of LTspice, but I don't consider that a good solution. 

I think the core issue is the hard coded paths to the location of the LTspice executable. Making this configurable in a way that works on Windows, mac and linux would be awesome.

akaeba, 2021-09-26:
Hello Bernienor,

on possible way can be to use a yml file configuration. The package 'os' can determine the running os:
```python
os.name  # returns running os
```
If you use it in a yml file:
```yml
nt: <Windows_Path>
posix: <linux_path>
```
In the pyLtSpice python script you load as first the yml confiuration, and then select teh active path
```python
# load config
fH = open(<ymlFile>, 'r+')  # open file for yaml loader
ymlConfig = yaml.load(fH, Loader=yaml.FullLoader)  # load condig
fH.close();                            # close file handle
# determine active path
actPath = ymlConfig[os.name] # selects os depending the right path
```
That's it :-)

bernienor, 2021-09-26:
Hi all,
I got around the problem with space in the path for the executable (in linux) by calling a local script:

`#!/bin/bash`
`wine /home/bernth/.wine/drive_c/Program\ Files/LTC/LTspiceXVII/XVIIx64.exe $@`

And changing the definition of LTspice in PyLTSpice/LTSpiceBatch.py:
`LTspice_exe = [r"./LTspice.sh"]`

This works with a few hickups.
When calling LTC.run() the filepath must be "./" Calling it with an absolute path causes LTspice to not find the file. This I think is a problem with LTspice and not PyLTSpice, as LTspice has no problem generating netlists when called with an absolute path.

I think a quick fix for closing this issue is to add a function to set the path to the LTspice executable, and to sugest using a shell script for calling wine with the executable when running windows.

A better solution could be to add a yml file with a set of default configurations for windows, mac and linux. But still have the possibility to let the user define a non standard path. But this solution requires more testing to ensure proper operation across operating systems.

nunobrum, 2021-09-26:
I think it justifies to introduce on the code some way to override the LTSpice exe location.

I'm proposing something of this kind

` setLTspiceRunCommand(r"./LTspice.sh") `

and also replacing the LTSpice_exe definition code by this, so in most cases it would be automatic
```
if sys.platform == "linux":
    LTspice_exe = 'wine C:\\\\Program\\ Files\\\\LTC\\\\LTspiceXVII\\\\XVIIx64.exe'
    LTspice_arg = {'run': ['-b', '-Run']}
elif sys.platform == "darwin":
    LTspice_exe = '/Applications/LTspice.app/Contents/MacOS/LTspice'
    LTspice_arg = {'run': ['-b']}
else:  # Windows
    LTspice_exe = [r"C:\Program Files\LTC\LTspiceXVII\XVIIx64.exe"]
    LTspice_arg = {'netlist': ['-netlist'], 'run': ['-b', '-Run']}
```
I haven't tested it myself since I don't have a linux machine, but, I remember Michael Wiebusch in his module was something of this kind and it was working.
**Can any of you with a Linux machine test whether this will work ?**

Also, I think it makes no sense to keep the LTSpiceIV_exe legacy. If someone is using this old version, he/she could use the command above to override it.

bernienor, 2021-09-27:
I have not been able to run wine directly from python under linux. I've tried any combination of forward and backward slashes I can think of, and some other tricks. Running it using a shell scrip works as long as I remove the absolute path for the files. This last problem has been reported to Analog Devices as I think it is a bug in LTspice.

01tot10, 2022-01-10:
Hello!

I'd like to join the wagon trying to get this thing running on Linux!

I had PyLTSpice running on my previous Ubuntu setup around March 2021, and I was able to run simulations and generate netlists, but it required me to do some dirty hacking to the codebase, LTSpiceBatch.py specifically, and I didn't reaaaaally understand what I was doing. I've since had to do a fresh install of my OS, and the time has finally come for me to start using PyLTSpice again.

I see things have changed a fair bit since last time, even up to the point that the LTCommander has been deprecated, and new and shiny SimCommander has appeared. Without changing things, trying to run `LTC = SimCommander("[FILENAME].asc")` is currently giving me:
`AssertionError: In this platform LTSpice doesn't have netlist generation capabilities`

Running the .asc-file in LTSpice to generate a netlist and running `LTC = SimCommander("[FILENAME].net")` works, but `LTC.run()` exits with:
```
Exception in thread sim1:
Traceback (most recent call last):
  File "/home/[USERNAME]/miniconda3/envs/python3/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/[USERNAME]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 157, in run
    cmd_run = LTspice_exe + LTspice_arg.get('run', '') + [self.netlist_file] + cmdline_switches
TypeError: can only concatenate str (not "list") to str
```
I feel like Linux is still not quite supported yet. Would the devs or anyone else be willing to collaborate / help me out here?
I want to feel hopeful, since at least last time I was able to get things going.

nunobrum, 2022-01-11:
I'm kind of limited in what respects OSX and Linux testing as I don't have a machine for it. 
From the error log that you posted I can see that the issue is on the code between lines 109 and 117 of the LTSpiceBatch.

There are some square brackets missing on the code.
```
if sys.platform == "linux":
    LTspice_exe = ['wine C:\\\\Program\\ Files\\\\LTC\\\\LTspiceXVII\\\\XVIIx64.exe']  # Added square brackets
    LTspice_arg = {'netlist': ['-netlist'], 'run': ['-b', '-Run']}
elif sys.platform == "darwin":
    LTspice_exe = ['/Applications/LTspice.app/Contents/MacOS/LTspice']  # Added square brackets
    LTspice_arg = {'run': ['-b']}
else:  # Windows
    LTspice_exe = [r"C:\Program Files\LTC\LTspiceXVII\XVIIx64.exe"]
    LTspice_arg = {'netlist': ['-netlist'], 'run': ['-b', '-Run']}
```
@01tot10 
Can you please try this and send me the result you'll get. Crossing my fingers that it will work.

01tot10, 2022-01-11:
> I'm kind of limited in what respects OSX and Linux testing as I don't have a machine for it. From the error log that you posted I can see that the issue is on the code between lines 109 and 117 of the LTSpiceBatch.
> 
> There are some square brackets missing on the code.
> 
> ```
> if sys.platform == "linux":
>     LTspice_exe = ['wine C:\\\\Program\\ Files\\\\LTC\\\\LTspiceXVII\\\\XVIIx64.exe']  # Added square brackets
>     LTspice_arg = {'netlist': ['-netlist'], 'run': ['-b', '-Run']}
> elif sys.platform == "darwin":
>     LTspice_exe = ['/Applications/LTspice.app/Contents/MacOS/LTspice']  # Added square brackets
>     LTspice_arg = {'run': ['-b']}
> else:  # Windows
>     LTspice_exe = [r"C:\Program Files\LTC\LTspiceXVII\XVIIx64.exe"]
>     LTspice_arg = {'netlist': ['-netlist'], 'run': ['-b', '-Run']}
> ```
> 
> @01tot10 Can you please try this and send me the result you'll get. Crossing my fingers that it will work. If it works I can commit this change to the master branch

Hey @nunobrum! Thanks for the quick reply! 

Not quite there yet, this is what I'm getting:

```
Tue Jan 11 11:00:32 2022 : Starting simulation 1
Exception in thread sim1:
Traceback (most recent call last):
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 166, in run
    self.retcode = run_function(cmd_run, timeout=self.timeout)
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 129, in run_function
    result = subprocess.run(command, timeout=timeout)
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'wine C:\\\\Program\\ Files\\\\LTC\\\\LTspiceXVII\\\\XVIIx64.exe'
```

And I can confirm, that no such file exists in my computer!
The location of my LTSpice-executable is the standard:
`/home/[USER]/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe`

also, running the following in terminal executes LTSpice correctly:
`wine '/home/[USER]/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe'`

nunobrum, 2022-01-11:
Can you please try changing line 110 to
    LTspice_exe = ["wine '/home/[USER]/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe'"]

01tot10, 2022-01-11:
@nunobrum. Yes, I would have also thought this should work, but not quite. The function call exits with:
```
Tue Jan 11 12:53:10 2022 : Starting simulation 1
Exception in thread sim1:
Traceback (most recent call last):
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 167, in run
    self.retcode = run_function(cmd_run, timeout=self.timeout)
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 130, in run_function
    result = subprocess.run(command, timeout=timeout)
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: "wine '/home/[USER]/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe'"
```
(I did replace [USER] with mine obviously)
Interestingly enough, copying the command between the strings ( `wine '/home/[USER]/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe'` ) from the exception above, and pasting and running it from terminal does work.

bernienor, 2022-01-11:
Hi all,

I think you could define LTspice_exe as follows:
LTspice_exe = ["/usr/bin/wine", "~/.wine/drive_c/Program
Files/LTC/LTspiceXVII/XVIIx64.exe"]

It looks like subprocess.call wants a list of a command and it's arguments,
so you need to separate the argument to wine, from the wine command.

Not sure why, but the process started by subprocess.call does not inherit
the environment from the shell. So calling ~/.wine... will not work, as
well as just calling 'wine' will not work, you need the absolute path. If
you use subprocess.run, you will not need the absolute path to wine. Just
need to figure out the absolute path to LTspice.

Bernie

01tot10, 2022-01-11:
Hi @bernienor! Thanks for joining in!

Nice, I think this definitely got us to places! LTSpice is getting invoked, although I'm getting encoding related errors with my source-file. Changing the encoding does make the whole thing simulate, but not quite as how I would expect. Also, changing the encoding of the source (naively with `iconv`) makes the simulation do weird stuff even when being ran the normal way from LTSpice standalone. I think I have to resolve this encoding thing first before going further.

Anyways, I think we are getting somewhere here! Thanks a lot @bernienor for your input!

01tot10, 2022-01-11:
Oh, just on a quick note, changing the encoding seems to change the µ-characters on the capacitors to something weird, which makes the thing not simulate properly, but having the µ-characters in there makes my Python script exit with:

```
    LTC = SimCommander(read_path + read_name)

  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/LTSpiceBatch.py", line 249, in __init__
    self.reset_netlist()

  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/SpiceEditor.py", line 666, in reset_netlist
    finished = self._add_lines(lines)

  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/site-packages/PyLTSpice/SpiceEditor.py", line 244, in _add_lines
    for line in line_iter:

  File "/home/[USER]/miniconda3/envs/python3/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb5 in position 267: invalid start byte
```
Anyways, I think I have to stop for today. I'll continue investigating tomorrow.

nunobrum, 2022-01-11:
Thanks @bernienor for the useful tips. 
You are right regarding the m and Meg. Not that sure regarding the micron symbol. I believe Analog Devices had the unfortunate idea of using it in some later versions of LT. 
In windows this is better managed, but I guess with wine it can create some issues. 
I think that replacing all microns per ‘u’ in the reset_netlist could do the trick. 
I’d like to integrate the changes @01tot10 did to make it work. Can you share your final code ?

01tot10, 2022-01-12:
Hey all!

@bernienor, Yes, one would think so, but it's not as simple as that. When I type in, say, 1uF to characterize one of my caps, the syntax gets converted to 1µF immediately when entered. And that character then has issues later down the chain... I guess this was introduced to LTSpice at one point, as suggested by @nunobrum. 

As a workaround, I changed the cap values containing microns to something based on nano, which seems to work. As such, I can confirm that LTSpice is currently working.

@nunobrum. I made a separate branch with my fixes, but it seems I can't push the branch into this repo. I'm not super experienced with git, but looking at the commit history, it seems like people have used forks previously. Or should I get a write access to branches other than main?

01tot10, 2022-01-12:
Allright, I made a branch in my local fork of the repo, where you can see the changes.
https://github.com/01tot10/PyLTSpice/commit/1b77a7b588c9cff22c84969504fd6719775bf2f6

As @bernienor mentioned, the path "~" doesn't get converted automatically to "/home/[USER]", so the path needs to be hard-coded into the script.. I see previously there has been some talk about some configuration shell-script, which could be a good idea.

Also, I needed to change the backslashes on setup.py to be able compile and install in Linux. I assume that would break the Windows compile then.

bernienor, 2022-01-12:
@nunobrum Right you are. I was not aware of the use of u in LTspice. Had not noticed :)

@01tot10 I've done more or less the same modifications to my code. However in setup.sh I only use a single forward slash '/'. And I think this also works in windows. 

As for the path to the home directory, this can be expanded by using os.path.expanduser('~').

`LTspice_exe = ["wine", (os.path.expanduser('~') + '/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe')]`

On my system (Ubuntu) the test scripts spawn several LTspice windows. I've not figured out how to eliminate this. I have to close the windows manually to get the scripts to finish. 

Cheers!
Bernie

01tot10, 2022-01-12:
@bernienor. Nice, that's a clever way to handle the home directory. Also, yes, those double forward slashes were a mistake in my part.. Thanks for notifying, and for your suggestions in general once again! 

I made a pull-request with the changes. It's not super extensively tested, but it compiles and runs in my Ubuntu distribution. But anyways it doesn't touch too many places so I'd imagine it to be fine.

nunobrum, 2022-01-12:
I've accepted the changes on the pull request

nunobrum, 2022-01-12:
Please free free to reopen in case anyone finds anything

01tot10, 2022-08-04:
Greetings from the Linux-land!

Everything's running as expected since more than half a year.

There's this one annoying wine-thing, that I've been trying to resolve every now and then, but since it's not a deal breaker I've always surrendered pretty quickly. This is more related to wine and not PyLTSpice, but I thought maybe it would still be appropriate asking for help here, in case someone has found a nice cure for this!

Basically, every time you're running simulations, wine is kind of polluting the Python terminal output with errors/warning related to it not being able to access all the parts of your filesystem. This results in errors looking like this, and happens with all simulations:

```
Running simulation 1...

Running threads
[<RunTask(sim1, started 139649394771712)>]

wine: Read access denied for device L"\\??\\Z:\\", FS volume label and serial are not available.
wine: Read access denied for device L"\\??\\Z:\\", FS volume label and serial are not available.
wine: Read access denied for device L"\\??\\Z:\\", FS volume label and serial are not available.
...
```

AFAIK this is due to the symlink `Z:` in wine pointing to `/`, which wine doesn't have privileges to (luckily).
I've been trying to get rid of the error via either removing the Z symlink with `winecfg`, or pointing it to somewhere within my homefolder, and while this gets rid of that specific error, I get a different one telling that wine is not able to access my netlists for some reason. The error message becomes:

```
Running threads
[<RunTask(sim1, started 140552578189056)>]
wine: could not open working directory L"PATH TO SIMULATION", starting in the Windows directory.
wine: could not open working directory L"PATH TO SIMULATION", starting in the Windows directory.
wine: could not open working directory L"PATH TO SIMULATION", starting in the Windows directory.
...
```

I've also tried to fiddle with the `WINEDEBUG` flag, trying to suppress all errors with `WINEDEBUG=-all`, but that also didn't do the trick. If I run LTSpice standalone (without PyLTSpice), I can supress the output via piping all the messages to /dev/null [*], but achieving that would require modifying PyLTSpice source and feels quite hacky.

[*] https://www.unix.com/linux/56325-wine-silent-mode.html

Would anyone have any clever ideas for this?

nunobrum, 2022-08-07:
Hi @01tot10 
I haven't touched Linux for more than a decade now. The wine/linux implementation was made with your help and all others on this thread. So, I'm probably not the best person to advise you on a cure for this. However, I'll try anyway. I have came up with three options that are completely hypothetical.

a) scrolling a bit up and reading the post of @bernienor from [26 Sept 2021](https://github.com/nunobrum/PyLTSpice/issues/35#issuecomment-927325143), we see that he was able to use a .sh script to call LTSpice.
Maybe there you can add the configurations that you need to "hide" the error.

b) Adding the pipe to /dev/null either using the setLTspiceRunCommand() or directly editing the code in line 110
```
if sys.platform == "linux":
    LTspice_exe = ["wine", os.path.expanduser("~") + "/.wine/drive_c/Program Files/LTC/LTspiceXVII/XVIIx64.exe"]
```
c) Finally, it seems to me that those errors may come up from definitions on the wine installation. Maybe the default home path or the order set on the path environment variable. Feel a bit shy suggesting this final one. I guess this is something you already looked at.

I hope this helps,
Nuno

closed 2022-09-11 by nunobrum