nunobrum/PyLTSpice #41: Changing Component Value to 0

m41o, 2021-12-07:
I discovered an issue when setting a component's value via SimCommander.set_component_value() to 0. Calculating log(0) is a bad idea...
It's a quick fix though and I've already done it in my dev environment. It's best to incorporate it into the main release though.

```
def format_eng(value):
    if value == 0.0:
        e = 0
        suffix = ''
    else:
        e = floor(log(abs(value), 1000))
        if -4 <= e < 0:
            suffix = "pnum"[e]
        elif e == 0:
            suffix = ''
        elif e == 1:
            suffix = "k"
        elif e == 2:
            suffix = 'Meg'
        else:
            return '{:E}'.format(value)

    return '{:g}{:}'.format(value* 1000**-e, suffix)
```

nunobrum, 2021-12-11:
Thanks for spotting this.
I will incorporate the change on the main dev.
Best Regards,
Nuno

nunobrum, 2021-12-11:
Change contained in https://pypi.org/project/PyLTSpice/1.6a0/

closed 2021-12-11 by nunobrum