nunobrum/PyLTSpice #42: Component Names not Starting with Correct Letter

m41o, 2021-12-07:
I am using your library to simulate a given schematic through Python. Some components in there do not follow naming conventions. E.g. one voltage source is called `Input_V_Bat`. In the netlist this results in a line starting with `V§Input_V_Bat`. Applying your regular expressions won't work in that case.

While users should probably follow naming conventions, I think it's best to incorporate also cases in which they don't like mine.

Changing

```
REPLACE_REGEXES = {
    'V': r"^(V\w+)(\s+\S+){2}\s+(?P<value>.*)$",
}
```
in `SpiceEditor.py` to `'V': r"^(V\S+)(\s+\S+){2}\s+(?P<value>.*)$",` did the trick. Not sure though if this is the best solution.

Thanks for your great work and cheers!

nunobrum, 2021-12-11:
Hi Markus,

I was really surprised with your finding. Decided to test with LTSpice, and indeed it adds the § sign, and also makes a lower case of the first v. 
I've made a few tests and saw that this behaviour happens with what ever components you use where you change the first letter. Resistors, Capacitors, Current Sources. So I have decided that the safest was to add a "§?" after every first letter. I think this will solve the issue and have low impact on the existing code.

I'll close this ticket once I have committed the files to Github and PyPi

nunobrum, 2021-12-11:
Change contained in https://pypi.org/project/PyLTSpice/1.6a0/

closed 2021-12-11 by nunobrum