nunobrum/PyLTSpice #44: Read Log Files in Operating Point

Eymay, 2022-01-29:
Not sure if it is possible as I didn't see it in the docs but I would like to see gm value of a mosfet in op simulation. The gm and more parameters are saved in .log file.

Support for Semiconductor Device Operating Points in the log file.
[image]

nunobrum, 2022-01-30:
Yes, it is possible. There is a module that does it. 
The https://github.com/nunobrum/PyLTSpice/blob/master/PyLTSpice/LTSpice_SemiDevOpReader.py
reads an operating point log file and reads all the information and returns the data in the form of a dictionary of dictionaries.
```
from PyLTSpice.LTSpice_SemiDevOpReader import opLogReader
data = opLogReader('simulation logfile.log')
gm = data['BSIM4 MOSFETS']['m16']['Gm']  # getting the GM value of transistor m16
mos_transistors =data['BSIM4 MOSFETS']
dict_with_all_gms = { ref : mos_transistors[ref]['Gm'] for ref in data mos_transistors }
```

nunobrum, 2022-02-08:
Seems to have worked. Closing the issue

closed 2022-02-08 by nunobrum