nunobrum/PyLTSpice #47: Licensing inconsistent across files

lpherr, 2022-03-01:
A variety of license terms are used across project files, including:
- Free
- lGPL v3
- General Public GNU License
- GPL v3
- blank

The intent appears to be a uniform GPL v3. Suggest all code file headers either:
1. state Licence: GPL v3 or similar according to existing convention
2. refer to the LICENSE file

nunobrum, 2022-03-05:
Thanks a lot for all your contributions. 
You did quite some work revising all the documentation. I really appreciate it. Now it's all neat and proper. 
Regarding the license issue, I am following your option 2.  I have placed in all "mature" files the reference to the LICENSE file. 
Just a quick question before I create the wheels for PIP installer. I would like to make a reference to you in the commit and on the version history. First of all, do you mind if I do it ? And second, how would you like the reference to be ? GIT username, Full Name, something else? email?
Just let me know.
Thanks once again,
Nuno

lpherr, 2022-03-09:
Thanks @nunobrum glad to help. You may reference me in the commit and version history. Please use Full Name, GIT username, and email (logan.herrera.github@gmail.com).

closed 2022-04-03 by nunobrum