nunobrum/PyLTSpice #50: Adding"µ" char in SpiceEditor regexes

pgepge, 2022-04-02:
Hello Nuno, 

The LTSpice schematic editor automatically replaces the "u" suffix (micro) with a "µ" char (ANSI/0xB5) if the Convert "µ" to "u" option in the Control Panel -> Netlist Options is deselected.
The SpiceEditor.py  REPLACE_REGXES currently do not scan for the "µ" char. Propose to add "µ" detection => (Meg|[kmuµnp]).

Thanks for sharing PyLTSpice!

lpherr, 2022-04-02:
Similarly it further appears femto- (f) and atto- (a) are not detected

nunobrum, 2022-04-02:
Hello !
I'm studying the code and it does seem that just adding the µ to the REGEX should do the trick.
There are a few improvements that I want to do, such as the get_component_value returning a float instead of just a string.
I'll commit a version that supports microns and femtos when I finish implementing this. Most probably tomorrow.

nunobrum, 2022-04-03:
Hello All!
I have just committed a version that supports also supports 'µ' for microns (besides u) and 'f' for femtos. 
I have also added a support function for the netlist manipulation called get_component_floatvalue() that takes in charge the task of converting from string to a float, taking into account the engineering 1000x qualifiers.

I have tested the new functions with a few testcases and all seems to be fine.

**Feel free to reopen this ticket in case you see something that is not right.**

fishyman1996, 2022-07-23:
Hello, related to this when using the µ for units the SImCommander doesnt run sims correctly. I think this error is being caused by the suffix not being interpreted when read correctly by reset_netlist in SpiceEditor.py line 941. Should be read in as "L1 N002 Vout 10µ" but being read in as "L1 N002 Vout 10�" instead

Test Code:
import os
from PyLTSpice.LTSpiceBatch import SimCommander
from PyLTSpice.LTSteps import LTSpiceLogReader
from PyLTSpice.LTSpice_RawRead import LTSpiceRawRead
import pdb
from matplotlib import pyplot as plt

LTC = SimCommander("Test.asc", parallel_sims=1)
LTC.run()
LTC.wait_completion()
LTR = LTSpiceRawRead("Test_1.raw")
print(LTR.get_trace_names())
y = LTR.get_trace("V(vout1)")
x = LTR.get_trace('time')  # Gets the time axis
plt.plot(x.get_time_axis(), y.get_wave(),'r')

plt.xlim([0,0.00002])
plt.show()

[image]

nunobrum, 2022-07-24:
Hello,

What is version that you are using?
I believe that this should be already solved in the latest version of PyLTSpice.
If this is not the case, please send me the your version and platform you are using.

Best Regards,
Nuno

fishyman1996, 2022-07-24:
Hello,

I believe I'm using 2.0.1 of PyLTSpice with python 3.6.2 on windows 10

Thanks

nunobrum, 2022-07-25:
Hello again,
That should normally work, unless your file encoding is not being supported.
Could you please try to find which encoding is the netlist being generated.
Please check this website that explains a way to do it.

https://softwareengineering.stackexchange.com/questions/187169/how-to-detect-the-encoding-of-a-file

If you are able to find the encoding, please try it directly on the detect_encoding.py on line 37.
Modify the line so that your encoding is listed first.
`for encoding in (<your encoding here>, 'utf-8', 'utf_16_le'):  # `

Let me know if you find the encoding that is working for you.
I might also help if you launch a command line and type "chcp" (gets the machine codepage setting) and give me the number that the command returns.

I hope this helps,
Best Regards,
Nuno

fishyman1996, 2022-07-27:
Hello,

Using notepad++ to get the encoding. The original .asc is ANSI. I added this to the detect_encoding.py and nothing changed. The net list that gets created becomes a UTF-8.

line 37 in detect_encoding.py doesn't get called, it goes to the else in line 41. If I replace line 45 with "return 'ANSI' " it reads µ correctly.

Is it worth nothing that I've had this issue on two different windows 10

Thanks

[image]

nunobrum, 2022-07-28:
I've just pushed a revision that tries the encodings in the following sequence ('utf_16_le', 'ansi', 'cp1252', 'cp1250', 'utf-8')
Also, I've accepted a pull request from @kgoodrick-rl which among other things it reads the full length of the file, thus reducing the possibility of mistakes.

fishyman1996, 2022-07-31:
Hello,

I've tried the new revision and I get this error in line 209 of SpiceEditor.py 
[image]

nunobrum, 2022-08-01:
Hello,

Thanks for your information.
This is unfortunately bad news for me, as it means that the current
approach to determine the encoding is not working well.
Thanks anyway for your cooperation,

Best regards,
Nuno

nunobrum, 2022-08-01:
Hello again,
just made yet some modifications on the detect_encoding function that hopefully will improve the situation with the encoding.
But, since this is a difficult problem to solve, I have added the encoding parameter on the SimCommander class initialization that allows to specify directly what encoding to use.
Best Regards,
Nuno

closed 2022-10-07 by nunobrum