nunobrum/PyLTSpice #53: LTSpice_RawWrite writing bad data for traces.

dardardario, 2022-06-15:
After downloading this package I tried testing it using the included test script here: [PyPi](https://pypi.org/project/PyLTSpice/) .
The script appears to error out with the following: 
`Found Numpy. Will be used for storing data
Traceback (most recent call last):
  File "./ltspice_RawWrite_test.py", line 5, in <module>
    tx = Trace('time', np.arange(0.0, 3e-3, 997E-11))
TypeError: __init__() missing 1 required positional argument: 'data'`

When adding the whattype arg to this (Test "real" and "voltage") the script runs and generates a .raw file which once openned up is just very large random pulses: 

[image]

dardardario, 2022-06-15:
Looks like the issue was not present in 0c3ecf0c9e945be23589ada94569c04cfb10cde4.

dardardario, 2022-06-15:
[image]
Could this be the root cause?

dardardario, 2022-06-15:
Rolling back to pip version 1.9 fixes the issue as well

nunobrum, 2022-06-15:
Ooops! Last minute changes...
I have just corrected this issue. It turns out that the numpy dtype was being changed from float32 to float64 when data was being set. This is why the routine stopped working.
Also, I have corrected the problem with the need of the whattype. Now it is an optional parameter and defaults to "voltage" for all non "time" and "frequency" traces.
Please sync to the latest commit and it should be working again. 

Note: This module is still quite embrionary and lacks the maturity in documentation and all the necessary data validations.
Thanks for understanding.

closed 2022-06-15 by nunobrum

nunobrum, 2022-06-15:
I've another issue to solve, so, I'll not propagate this to PyPy (pip) repository right away. 
For a quick fix, you can try updating these lines at the Trace __init__ function

```
        DataSet.__init__(self, name, whattype, len(data), numerical_type=numerical_type)
        if USE_NNUMPY and isinstance(data, (list, tuple)):
            self.data = array(data, dtype=self.data.dtype)
        else:
            self.data[:] = data[:]  # This way the dtype is kept
```