nunobrum/PyLTSpice #56: Workaround on terminating simulation

WKNGAP, 2022-08-04:
I found terminating simulation comes in handy when operating with a large batch of simulations and it seems to me that this library doesn't provide a way for me to do this.

Here is the modified example code provided by PyLTSpice which will terminate the simulation when it take more then 10 seconds to run.
```

import os
from PyLTSpice.LTSpiceBatch import SimCommander

#library needed to termaiate LTspice in windows
from time import sleep
import psutil

#Function to termaiate LTspice in windows
def KillLTspice():
    PROCNAME = "XVIIx64.exe"

    for proc in psutil.process_iter():
        # check whether the process name matches
        
        if proc.name() == PROCNAME:
            print("killing ltspice")
            proc.kill()
#end function

def processing_data(raw_file, log_file):
    print("Handling the simulation data of %s, log file %s" % (raw_file, log_file))

# select spice model
LTC = SimCommander("Batch_Test.asc")
# set default arguments
LTC.set_parameters(res=0, cap=100e-6)
LTC.set_component_value('R2', '2k')
LTC.set_component_value('R1', '4k')
LTC.set_element_model('V3', "SINE(0 1 3k 0 0 0)")
# define simulation
LTC.add_instructions(
    "; Simulation settings",
    ".param run = 0"
)

LTC.reset_netlist()
LTC.add_instructions(
    "; Simulation settings",
    ".ac dec 30 10 1Meg",
    ".meas AC Gain MAX mag(V(out)) ; find the peak response and call it ""Gain""",
    ".meas AC Fcut TRIG mag(V(out))=Gain/sqrt(2) FALL=last"
)

LTC.run()
#LTC.wait_completion() # this line is no longer needed
timeout=10 #10 seconds timeout used to test out this function
TimeoutCount=0
#The code below are mainly copy form wait_completion() function
LTC.updated_stats()
while len(LTC.threads) > 0 and TimeoutCount<timeout: #added stop condition: break the loop after timeout
    TimeoutCount+=1 # line added to achieve this function
    sleep(1)
    LTC.updated_stats()
	
KillLTspice()# call termaiation function, will do nothing if the simulation exit in time.

# Sim Statistics
print('Successful/Total Simulations: ' + str(LTC.okSim) + '/' + str(LTC.runno))
```

Holp you find this code useful and may become on of the functions of PyLTspice.

nunobrum, 2022-08-06:
Thanks for this. 
I will integrate this in a later version  of PyLTSpice.
I'll close this issue when it will be implemented and tested.
Wishing you a nice summer ahead,
Nuno

nunobrum, 2022-09-07:
Hello @WKNGAP 

I have tentatively integrated your code as a staticmethod inside the SimCommander class and implemented in the wait_completion an abort_all_timeout that will in last resort kill all LTSpices
 ```
    @staticmethod
    def kill_all_ltspice():
        """Function to terminate LTSpice in windows"""
        import psutil
        for proc in psutil.process_iter():
            # check whether the process name matches

            if proc.name() == PROCNAME:
                print("killing ltspice")
                proc.kill()

    def wait_completion(self, abort_all_timeout=0) -> bool:
        """
        This function will wait for the execution of all scheduled simulations to complete.

        :param abort_all_timeout: this is an abort all timeout. This timeout is reset everytime that a simulation is
                                  completed. It should be noted that this can override the timeout defined on the
                                  SimCommander class. The difference is that the SimCommander instance timeout property
                                  is implemented by the subprocess class, and the abort_all_timeout is a last resort
                                  timeout that will kill all LTSpice processes, even those that were not spawned by
                                  SimCommander.
        :type abort_all_timeout: int
        :returns: True if all simulations were executed successfully
        :rtype: bool
        """
        self.updated_stats()
        timeout_counter = 0
        sim_counters = (self.okSim, self.failSim)

        while len(self.threads) > 0:
            sleep(1)
            self.updated_stats()
            if abort_all_timeout:
                if sim_counters == (self.okSim, self.failSim):
                    timeout_counter += 1
                else:
                    timeout_counter = 0

                if timeout_counter > abort_all_timeout:
                    self.kill_all_ltspice()
                    return False

        return self.failSim == 0
```
I note that there was already a timeout mechanism implemented directly by the subprocess. So, this is an alternative way of doing it.

I will commit this implementation as soon as I test it a bit better.
Let me know if this works for you or if you have any other suggestions.

Best Regards,
Nuno

nunobrum, 2022-09-11:
Hi @WKNGAP 

I've added a kill_all_ltspice function as a classmethod.
Also added the possibility of having a timeout on the wait_completion.
I hope this maps what you were suggesting.
Best Regards,
Nuno

closed 2022-09-11 by nunobrum