nunobrum/PyLTSpice #59: SemiDevOpReader not functional on MacOS

ArthurAdmiraal, 2022-09-10:
Heya, thanks for creating this awesome tool! I'm having some trouble with the SemiDevOpReader on MacOS, which I believe is due to no encoding being selected.

### The issue
An LTSpice simulation generated the log file `Mos.log` on MacOS. Running `PyLTSpice.LTSpice_SemiDevOpReader.opLogReader("Mos.log")` returns an empty dictionary `{}` rather than the expected device parameters.

### Suggested fix
Comparing the implementation of `opLogReader` with `LTSpiceLogReader`, the encoding detection is missing. This would cause all sorts of mayhem in parsing the file.

Hope this helps!

nunobrum, 2022-09-11:
Hi @ArthurAdmiraal 

I've just added the detect_encoding to the opLogReader
Let me know if it solves the problem.

Thanks for the fix suggestion.

Best Regards,
Nuno

ArthurAdmiraal, 2022-09-13:
Version 2.0.2 does indeed solve the problem, thanks taking the time to look into this!

closed 2022-09-13 by ArthurAdmiraal