nunobrum/PyLTSpice #67: Problems parsing parameters of a subckt

DorfLurch, 2022-12-08:
When reading a *.net file using SpiceEditor(), parameters given to a subcircuit are not correctly recognized. 

The line on a netlist
`XX1 0 N001 sub_test params: Res = 1`
 results into an malformated result of get_component_info('XX1'):
`{'designator': 'XX1', 'nodes': ' 0 N001 sub_test params: Res =', 'value': '1', 'line': 1}`

An example is attached ti the issue.

[test.net.txt](https://github.com/nunobrum/PyLTSpice/files/10187017/test.net.txt)

nunobrum, 2022-12-13:
@DorfLurch 
Thanks for reporting this issue.

I have a quick fix for your problem. Can you please find the SpiceEditor.py file on your python installation and replace the line 91 with the one below. 
Please let me know if this works well for you. I'll also run a few tests on my side to make sure that it cover other cases.
`    'X': r"^(?P<designator>X§?\w+)(?P<nodes>(\s+\S+){1,99}?)\s+(?P<value>\w+)(\s+params:)?(?P<params>\s+\w+\s*=\s*\S+)*\s*$",  # Sub-circuit, Parameter substitution not supported
`
If all goes well, I'll commit this to the repository quite soon.

Best Regards,
Nuno

nunobrum, 2022-12-28:
Hi there, 

sorry for the late reply.  I've been distracted with other issues.
I came up with something that doesn't ignore the other parameters. Till I find something better you can try.
`'X' : "^(?P<designator>X§?\w+)(?P<nodes>(\s+\S+){1,99}?)\s+(?P<value>\w+)(\s+params:)?(?P<params>(\s+\w+\s*=\s*[\d\w{}()\-\+\*\/]+)*)\s*\\?$"

Let me know if that solves the problem.

BTW, you may want to try the no_ltspice_prefix branch, where I've already included this fix, and also, makes the PyLTSpice more user friendly, without having to prefix all functions with LTSpice.

Best Regards,
Nuno

nunobrum, 2023-01-02:
Integrated in the last update.

closed 2023-01-02 by nunobrum