nunobrum/PyLTSpice #80: Make SimCommander work on systems with non-standard installation of LTSpice

basejumpa, 2023-01-10:
In order to use SimCommander on systems with non-standard installation of LTSpice there's already method Method `setLTspiceRunCommand`.

But unfortunately this cannot be used. It's kind of hidden by the constructor `__init__`. (Currently class `SimCommander` defines only that one constructor.) It is because the constructor, so during object creation, already uses the internal variable `LTspice_exe` and tries to create a process from it. That is BEFORE the user can change the executable's name via a call of the object's function `setLTspiceRunCommand`.

There are different ways how to solve it. I'd like not to suggest any.

Here I'd like to publish my WORKAROUND how to create a code which runs on different machines with different kinds of installations. As soon there's any PyLTSpice version available at https://pypi.org/project/PyLTSpice/ I'll remove my workaround.

``` python
# WORKAROUND because of not-working function setLTspiceRunCommand() of PyLTSpice 2,3, 3.0

from typing import Union
from shutil import which

def getLTspiceRunCommand():
    """Find LTSpice installation on system (in PATH variable)"""
    run_commands_to_check = [
        'XVIIx64',                 # 64-bit installer for windows from https://www.analog.com/en/design-center/design-tools-and-calculators/ltspice-simulator.html
        'ADI LTspice XVII',        # Scoop package manager via bucket https://github.com/SayCV/scoop-cvp
    ]
    return next(c for c in run_commands_to_check if which(c))

run_command = getLTspiceRunCommand()

if not run_command:
    print("ERROR: No LTSpice found on system. Please install it.") 

# WORKAROUND for the problem that setLTspiceRunCommand cannot be used as long as default installation path is not present on the system. (in parallel to possible others). Reason: The Constructor of SimCommander executes the LTSpice executable. This is BEFORE the LTSpice executable name can be set from user via call of method setLTspiceRunCommand:

class SimCommander_setLTspiceRunCommand(SimCommander):
    
    def __init__(self, run_command: Union[str, list]):
            super().setLTspiceRunCommand(run_command)

SimCommander_setLTspiceRunCommand(run_command)

# END OF WORKAROUND

# Now constructor uses variable LTspice_exe pointing to existing executable.
LTC = SimCommander(schematic.asc')
```

nunobrum, 2023-01-15:
Hi there,

thanks for your contribution. I see what you are saying, but, it is strange that it is not working
The LTspice_exe is a variable of the sim_batch.py module but, by declaring it global inside of the setLTspiceRunCommand, it was able to update it. At least in earlier Python versions.

If this is not working, I think best that LTspice_exe would be part of the SimCommander class. This way it allows that different instances of SimCommander can have different spice tools.
I haven't updated this in the revision I did today, as this is going to be completely revamped. So to support having a simulation server machines.
Best Regards,
Nuno

nunobrum, 2023-01-24:
I've updated the way the LTSpice tool is detected and the way it can be manually specified. 
It's online from version 3.1

closed 2023-01-24 by nunobrum