nunobrum/PyLTSpice #88: Spicelines misinterpreted in get_component_info

dbriscoe511, 2023-03-08:
Spicelines that are added to a part ate not correctly interpreted by the get_component_info method. On some items such as voltage sources, the spice lines are added to the value, on other parts they are ignored. 

Use case: pulling a bunch of tolerances for a worst case analysis

example code:
`
# select spice model
LTC = SimCommander("voltage_divider.asc")
# set default arguments
LTC.set_parameters(res=0, cap=1e-15)
#get the parts
part_list = LTC.get_components()
for part in part_list:
    #get the part info
    part_info = LTC.get_component_info(part)
    print(f"{part} value is {part_info}")
`

Output:
`
R1 value is {'designator': 'R1', 'nodes': ' N001 out', 'model': None, 'value': '1k', 'line': 1}
R2 value is {'designator': 'R2', 'nodes': ' out 0', 'model': None, 'value': '1k', 'line': 2}
C1 value is {'designator': 'C1', 'nodes': ' out 0', 'model': None, 'value': '1µ', 'line': 3}
V1 value is {'designator': 'V1', 'nodes': ' N001 0', 'value': '5 tol=1', 'line': 4}
`

Sample circuit:
voltage_divider.asc
`
Version 4
SHEET 1 880 680
WIRE 160 32 -16 32
WIRE 160 64 160 32
WIRE -16 96 -16 32
WIRE 160 160 160 144
WIRE 192 160 160 160
WIRE 240 160 192 160
WIRE -16 192 -16 176
WIRE 240 240 240 224
WIRE 240 240 160 240
WIRE 160 272 160 240
FLAG 160 272 0
FLAG -16 192 0
FLAG 192 160 out
SYMBOL res 144 48 R0
WINDOW 0 -42 46 Left 2
WINDOW 3 -41 76 Left 2
WINDOW 39 -57 99 Left 2
SYMATTR InstName R1
SYMATTR Value 1k
SYMATTR SpiceLine tol=1
SYMBOL res 144 144 R0
WINDOW 0 -44 51 Left 2
WINDOW 3 -42 74 Left 2
WINDOW 39 -62 95 Left 2
SYMATTR InstName R2
SYMATTR Value 1k
SYMATTR SpiceLine tol=1
SYMBOL cap 224 160 R0
WINDOW 39 24 84 Left 2
SYMATTR InstName C1
SYMATTR Value 1�
SYMATTR SpiceLine tol=10
SYMBOL voltage -16 80 R0
WINDOW 39 24 124 Left 2
SYMATTR InstName V1
SYMATTR Value 5
SYMATTR SpiceLine tol=1
`

dbriscoe511, 2023-03-15:
1.But, what puzzles me in your example, is that tol is not really a parameter accepted by LTSpice nor any other spice implementation.
    It is a default parameter for resistors (you can view it by right clicking on the part), although admittedly it is not used in the simulations.
    Spice recognizes this as a "parameter" of other components and will give an "unknown parameter" error on parts that are not resistors. In this case tol is just the parameter I had issues with, but I use the parameters/spice-lines of parts somewhat frequently when customizing their behavior. I realize this is probably a nightmare to get working with Regex. in the below schematic you can see where this info is, you can get it by ctrl-right clicking.  
[image]
2. a toolkit for making worst case analysis and monte-carlo
    This is what I was implementing, and I was planning on making a contribution once I got things working. I realize that using a "tol" field in this application might not be the best way to do it, but it seemed like a good starting point to me since it is a default parameter of resistors. 

I have been using the article you mentioned but I was moving to python since default LT spice WCA has quite a few major limitations

- If a run causes your circuit to break, there is no easy way to look at what component values caused the issue
- The output file sizes are insanely large for complex circuits since all of the data is saved. frequently, I only want a measurement or two from each run, but end up with a 100GB file for that. using python, it would be nice to only save the full data for a run if a special condition was detected, such as a voltage exceeding a limit. 
- It is pretty labor intensive to set up a WCA run using the methods above, and if something goes wrong half way through the simulation you cannot restart it. 

I appreciate the work on the project, and if you have a suggestion for a better way to implement a WCA toolbox, let me know.

nunobrum, 2023-03-15:
Hi @dbriscoe511, I hear your and I fully understand the drawbacks you are describing. 
The WCA and Monte-Carlo toolkit is still under conception, with @mvanriet that volunteered to work on that part.
I think the first approach will be to avoid the labor intensive part of setting up the WC and MC functions on each component (resistor/capacitor/opamp). This of course doesn't solve completely the problem of simulation sizes.
Second part would be to divide simulation into smaller chunks, and then have the tools to merge the results.

Unfortunately, until this work is done, I haven't a better way to do things.  
You see. All of this is done as a hobby in spare time. 
The only incentive are the nice comments and likes that I receive.  Progress is slow, but steady.
Maybe in a couple of months we might have something ready to be used.

dbriscoe511, 2023-03-16:
Alright, since the more interesting discussion is about this monte carlo issue, ill move it over to the discussion on this topic. I have some free time to contribute to this project but would like to make sure how I was thinking this could be implemented makes sense.

closed 2023-03-16 by dbriscoe511