nunobrum/PyLTSpice #98: Read measurement results from log file after simulating with simrunner

MoritzSeidel-2000, 2023-05-16:
I would like to start a Simulation via SimRunner and get a signal via the .MEAS-Function in my .LOG -file
When simulating the LT-Spice files via the "SimRunner", the "Meas-Instructions" are taken over into the newly created netlist, but the result is not displayed in the log file. Instead a "Measurement failed" message is displayed.
Also via the "add_instructions" function no result of the "Meas" function can be displayed in the log file.

nunobrum, 2023-05-16:
Most likely the measurement is failing because it has a syntax error.
Please make sure to test the measurement command directly in LTSpice before
adding it through the add_instruction method.
Add the .meas instructions with the S key and type .MEAS
You can test the .meas instructions in LTSpice by right clicking on them.

MoritzSeidel-2000, 2023-05-16:
thanks for the quick reply :)
The problem unfortunately still exists
The "Meas-Command" within LT-Spice works fine as long as LT-Spice is started by hand.
As soon as the simulation is started via the "SimRunner" the "Measurement Error" occurs in the log file. Thereby the Meas-Command was correctly taken over into the generated Netlist.

MoritzSeidel-2000, 2023-05-23:
I was able to solve the problem with the "Meas-Instructions".
The problem was not with PyLTSpice itself. It was just not possible to measure the drain current directly at the transistor, probably because it is an external model. With a 0V voltage source at the drain connection the current can now be determined without problems :D

closed 2023-05-23 by MoritzSeidel-2000