nunobrum/PyLTSpice #107: Simulation timesteps are larger than they should be?

person678, 2023-07-25:
I am simulating a circuit which has a sample and hold block reading from a PWM input, which inputs a new value every 30us. I use a pulse generator on the S&H block to sample from the PWM at the same rate, every 30us. 

When I simulate this in the GUI for say 18s, my timesteps are nice and small. For example, around 1s in: 
Time                                         Voltage
9.999917944850031e-01	4.856268e-01
9.999921310392923e-01	4.857006e-01
9.999928041478707e-01	4.858485e-01
9.999941503650277e-01	4.861465e-01
9.999968427993416e-01	4.867493e-01
1.000001115131372e+00	4.877184e-01
1.000002615131372e+00	4.880630e-01
1.000003500000000e+00	4.882686e-01
1.000004730262743e+00	4.885558e-01
1.000005539474514e+00	4.887461e-01
1.000006500000000e+00	4.889728e-01

And in total, for 18.05s there are over 27 million timesteps. 

However when I simulate the exact same circuit from PyLTSpice, it seems to be simulating at a much lower resolution. 
At 1s, the time intervals are: 
Time                            Voltage
0.9962401379999999,          -0.0673556
0.9972615533999999,          -0.06735497
0.998283155400001,            -0.06735147
0.9993124307999788           -0.06735235
1.00033665,                          -0.067355804
1.00135806                           -0.067354634
1.0023798599999998           -0.06735124
1.0034086199999988,          -0.06735268
1.0044309660000006           -0.067355946
1.0054518737999996           -0.06735442

As you can see, the timesteps are much larger (and the voltages are different). In total there are far fewer timesteps, around 18k. 
I've verified the input PWM files to the circuits are correct and identical, and I'm loading in the correct netlist (it's being generated from the exact same asc file). I'm adding the exact same simulation command to SpiceEditor. 

It's possible, I am making an error in my setup, but I'm struggling to spot it if so. Any help is appreciated. 

[MackeyGlassSim.txt](https://github.com/nunobrum/PyLTSpice/files/12160938/MackeyGlassSim.txt)

nunobrum, 2023-07-26:
Hi @person678 

In order to diagnose this issue I would propose some tests. 

First I would like for you to compare the size of the generated raw files. That establishes where the issue is on the simulation and not on the reading of data. 

Secondly, remove any editions to the netlist, to be sure that we have exactly the same simulation. LTSPICE doesn’t make fixed step simulations. It adjusts the time step dynamically during simulation. Simulations with more noise will require smaller time steps. 
Then you can force a fixed time step on the .TRAN command. See help file. 

Then we need to make sure we are calling the same LTSpice install and it is receiving the same ini file. There are some differences on the between the new and older LTSpice installations. And ini files can change the tolerances and make the simulation faster. 

I hope this helps. 

Nuno

person678, 2023-07-27:
By comparing the raw files and then the netlists I realised that the input was not being set correctly - I had accidentally deleted the neccessary "PWL file=" bit from the string specifying the file location at some point. Corrected that and all working fine now. Thank you very much for your help and apologies for taking up your time!

Paul

person678, 2023-07-28:
Apologies again, I thought I'd fixed this but having implemented that fix, I compared the RAW files and the PyLTSpice does indeed seem to be solving with reduced resolution. The PyLTSpice simulation RAW file is on the left, the GUI simulation on the right. 

I've selected show datapoints, and you can see the PyLTSpice one is missing detail in the waveform due to the reduced timesteps. 
I've verified that it seems to be running the same version of LTSpice. 

[image]

Here I have used the TRAN command to force a maximum timestep of 0.1us to see if that would inprove things. (I also tried a less extreme value of 7.5us). I've marked data points on this RAW, and the grid markers are every 20us. As you can see, the datapoints are far less frequent than every 0.1us. This is causing significant error when I analyse my results. 

[image]

Any help immensly appreciated.

nunobrum, 2023-07-28:
This is a quite puzzling situation. You know that the only thing that PyLTSpice does is create a netlist and then call LTSpice to run it.
So, I would like for you to do the following:

1. In LTspice, run a simulation of a .asc file and then rename the output.raw file to original.raw
2.  Generate the spice netlist manually (Tools -> Generate netlist)
3. Open a command prompt terminal (write "cmd" in search windows and enter)
4. Navigate to the folder where you have your .net file (cd "path_name" in command prompt and enter)
5. Launch the command line simulation manually, which is  the same what PyLTspice does. (call C:\Users\nunob\AppData\Local\Programs\ADI\LTspice\LTspice.exe -Run -b <name_of_net>) Update with your username.
6. Compare the raw file generate with the original.raw. If all went well, they should have the same size. If they don't, this is some issue that is not related with PyLTspice. If the have the same size, continue to the next step.
7. Compare the netlist generated with LTspice with the one generated with the PyLTSpice. At this point, we can gradually introduce the updates made by the PyLTspice manually, and each time, call the command issued in step 5. Surely, one of the updates is causing the change in timestep. It's just a question of determining which one it is.

PyLTspice, is a convenience tool to automate the updates of the netlist and call a simulation. It's just a driver, all happens with LTSpice. For some reason, (different models, different simulation settings, different stimuli), it is causing the simulation to speed up.

I hope this helps,
Nuno

nunobrum, 2023-07-28:
Just another thing.
```
PyLTspice searches for an LTspice installation in the following sequence:
        for exe in (  # Placed in order of preference. The first to be found will be used.
                os.path.expanduser(r"~\AppData\Local\Programs\ADI\LTspice\LTspice.exe"),
                r"C:\Program Files\ADI\LTspice\LTspice.exe",
                r"C:\Program Files\LTC\LTspiceXVII\XVIIx64.exe",
                r"C:\Program Files (x86)\LTC\LTspiceIV\scad3.exe",  # Legacy LTspice IV
        ):

```
This means that if you have more than one LTspice installation it can happen that the one that is being used by PyLTspice is not the same as the one you are using.

person678, 2023-08-11:
Hi, 

Appreciate your detailed response once again. I'm still struggling to correct the issue after some investigation, but I do agree the issue likely does not like with PyLTSpice and rather my system, so I will close this issue.

closed 2023-08-11 by person678

nunobrum, 2023-08-11:
If you can send me your test bench I can take a look at it.
It’s the least I can do.