nunobrum/PyLTSpice #112: Apply Control

Wuerike, 2023-08-13:
Hi!

Is it possible to interact with the circuit during the simulation?

Im researching on MPPT AI based control, so Im looking a manner to be able to simulate a circuit in python and for each time step, read some values and act on control.

nunobrum, 2023-08-13:
Hi!

Just to understand what you mean by time step. Do you mean:
1) to interact with the simulation results while the simulation is still running, or
2) Look at the simulation results after each .STEP execution.

If you mean 1), there is no direct way of interacting with a running simulation. 
What you can try to do, is to make use of .LOADBIAS and .SAVEBIAS primitives and ask the simulation of just a small time step.
You'd use the .SAVEBIAS at the end of your time-step, and use the .LOADBIAS at t=0 to continue from the previous simulation.
The .LOADBIAS and .SAVEBIAS syntax is described here.
https://ltwiki.org/files/LTspiceHelp.chm.html
Another option, maybe a bit more straightforward but more costly in terms of processing time, is to make increase time runs by modifying the .TRAN <stoptime> primitive.
All of the above is possible using the SpiceEditor or AscEditor for manipulating the .LOADBIAS / .SAVEBIAS or .TRAN commands and then use the SimRunner to make the simulation calls, and RawRead/LogRead to inspect the simulation results.

For 2) is really easy, as you can select which .STEP you want to read on the RawRead Class.

I hope this helps,
Nuno

Wuerike, 2023-08-14:
Thanks Nuno!

I referred about 1, interact with the simulation results while the simulation is still running.

I'll try your ideas!

nunobrum, 2023-08-26:
Closing this issue as it is unrelated with the PyLTSpice. 
For this kind of questions, you can use the Discussions forum.

closed 2023-08-26 by nunobrum