nunobrum/PyLTSpice #115: [Feature Request] Add Support for Q-Spice

ard0gg, 2023-08-24:
Mike Engelhardt, the creator of LTSpice, has recently release a new software called [QSpice](https://www.qorvo.com/design-hub/design-tools/interactive/qspice/). I haven't looked much into it, but I suspect there are similar data structures to LTSpice and may be relatively easy to add support for.

nunobrum, 2023-08-26:
Thanks for the heads up. I've just pushed into the dev branch a revised version that supports the QSPICE.

In this version there is a class that represents the QSPICE simulator, and the RawRead class had a minor update to support the new qraw format.

Presently the output raw file is forced to have the extension .raw to minimize the impact on existing code.
It works, but this is something that I have to change in the future, because the qraw files are not compatible with the LTspice raw files.
With QSPICE qraw, as all variable is stored with double precision. It is not possible to use LTSpice to open qraw files and QSPICE to open .raw files.

On the examples folder, you can find one example that uses the QSPICE to make a simulation.

Please let me know if you have some issues with it. If not, I can pull this to the main branch.

nunobrum, 2023-09-21:
This is implemented on the sister library spicelib. 
Closing this ticket here.

closed 2023-09-21 by nunobrum