nunobrum/PyLTSpice #116: Change since 4.0.6: .raw file not found

patzf, 2023-08-27:
I basically use the library in a very basics way

```
import logging
import sys
from PyLTSpice import SimRunner
from PyLTSpice import SpiceEditor
import PyLTSpice
PyLTSpice.set_log_level(logging.DEBUG)

ascFilePath = sys.argv[1]

# select spice model
LTC = SimRunner()
LTC.create_netlist(ascFilePath + '.asc') 
netlist = SpiceEditor(ascFilePath + '.net')

# set default arguments
netlist.set_parameters(duty=42, freq=100e3)

# define simulation
netlist.add_instructions(
    "; Simulation settings",
    ".tran 1000u"
)

LTC.run(netlist, run_filename=ascFilePath) #without run_filename, a _1 is added at the end of the generated files
```

I use the above code and then, in  second step run a python module that parses the data out from the .raw file and does some printing./plotting

This works fine until V 4.0.6.
For newer versions it seems that the .raw file is deleted right after simulation?

nunobrum, 2023-08-27:
This is really strange. The simulation files are only erased if you explicitely call the file_cleanup() method int the SimRunner.
Could it be that the raw file was not even created because of some problem in the execution of the simulation.
There was quite some refactoring done from the 4.0.6 to the 4.1.0.

asereq, 2023-08-31:
Same here, run_filename does not work.

I found that it changes (in the wine command, linux), the netlist file name argument by the run_filename  parameter.

But I would like to be clear: this project is great! Thanks so much for your work!

patzf, 2023-08-31:
> This is really strange. The simulation files are only erased if you explicitely call the file_cleanup() method int the SimRunner.
> 
> Could it be that the raw file was not even created because of some problem in the execution of the simulation.
> 
> There was quite some refactoring done from the 4.0.6 to the 4.1.0.
> 
> 
> 
> Can you send me the logs ?

I will provide some logs soon 😎

nunobrum, 2023-09-01:
I've tested your code with and without the run_filename parameter. 

What I found is that when you use the run_filename parameter you have to be consistent with the extension you are adding.
When declaring the netlist you use .'net extension

`netlist = SpiceEditor(ascFilePath + '.net')`

and when you run, you are not providing the extension.

`LTC.run(netlist, run_filename=ascFilePath)`

If you use :
`LTC.run(netlist, run_filename=ascFilePath+'.net')`
it works.

At least in my installation with the both 4.1.0 and 4.1.1 version. Please let me know if this works for you too.

patzf, 2023-09-03:
> At least in my installation with the both 4.1.0 and 4.1.1 version. Please let me know if this works for you too.
Hi @nunobrum : Well, I just got the time to start my machine and wanted to grab some logs. But the info in the lat reply seems indeed solved the problem. I would not hve thought of the extension to be the problem, but yes, it was of course inconsistent

Thanks for resolving the issue!

closed 2023-09-03 by patzf