nunobrum/PyLTSpice #118: PyLTSpice freezes when creating a net list from

Yaafteh, 2023-09-03:
When running the following code in mac  (MacBook M1 chip, not sure it matters) with Python3.11 and PyLTSpice 4.11 (tried previous versions as well, which lead to the same results) the code freezes when it tries to create the netlist file (runner.create_netlist command). PyLTSpice works when 'filename.net' it passes to the code directly (pre-created .net file).

fn = '/path/filename.asc'
runner = SimRunner(output_folder='/path/', verbose=True, timeout=3)
runner.create_netlist('filename.asc')

nunobrum, 2023-09-03:
As far as I know the Mac OSX version of LTspice doesn't support the creation of netlists using the command line.
At least this was the case the last time that I've tried it on a Mac. To be honest, it was a long time ago.
Can you try launching a terminal, confirming the LTSpice App location and issue the following command.

`/Applications/LTspice.app/Contents/MacOS/LTspice -netlist filename.asc`

If this creates the `filename.net` then this is a problem with PyLTSpice which I can try to solve. If not, there isn't anything I can do.

Yaafteh, 2023-09-03:
Yes, you are right! I tried it on my side. It doesn't generate the netlist in terminal as well. Thanks for prompt response and creating this tool.