nunobrum/PyLTSpice #119: Passing more arguments to callback function

Yaafteh, 2023-09-06:
Not sure if there is a way to pass more arguments to post processing function in run command or not. 
If yes, please advice. It there a workaround?
If not it would be ideal to add this feature to pass more arguments to run function, since it gets only the function handler, as far as I know, more args cannot be passed to post processing function.

It would be ideal if I could pass kwargs to the processing function. like:
runner.run(netlist, run_filename=fn_netlist_temp, callback=processing_data, callback_kwargs = {'R': 1, 'C':2})

nunobrum, 2023-09-07:
Hi @Yaafteh , I agree this is a good idea. 
I think this implemention does require change the sim_runner and the run_task. I can't imagine any other more elegant way of doing it.
I'll add this in the next version.
Just another thing completely out of topic. I'm moving most of the code to the [spicelib](https://github.com/nunobrum/spicelib) repository and then scale down the PyLTspice to use the spicelib as resource.

[https://github.com/nunobrum/spicelib](https://github.com/nunobrum/spicelib)

nunobrum, 2023-09-08:
I've just pushed into [spicelib]( https://github.com/nunobrum/spicelib/tree/main) a version that supports the callback arguments.
Please let me know if it matches your expectations.
I highly recomend that you use spicelib for the time being. 
PyLTspice will have some major redesign in the near future to be based on the spicelib.

closed 2023-09-08 by nunobrum