nunobrum/PyLTSpice #120: Documentation confusing about 'Sim Analysis Toolkit'

xgzi, 2023-09-11:
Hi,
I'm sure you want the user to manipulate the ASC file when utilizing the "Sim Analysis Toolkit" for "MonteCarlo" or "WorstCase" simulation.
However, I found in your documentation and its example codes, that the ".net" or ".asc" suffix is misleading.

> ```python
> sallenkey = AscEditor("./testfiles/sallenkey.asc")  # Reads the asc file into memory
> mc = Montecarlo(sallenkey)  # Instantiates the Montecarlo class, with the asc file already in memory
> # Finally the netlist is saved to a file
> mc.save_netlist('./testfiles/sallenkey_mc.net')
> ```
> 
> When opening the created sallenkey_wc.net file, we can see that the following circuit.

Since the editor here is `AscEditor`, I referred to its `write_netlist` method as below:

> In `asc_editor.py`
> 
> ```python
>     def write_netlist(self, run_netlist_file: Union[str, Path]) -> None:
>         if isinstance(run_netlist_file, str):
>             run_netlist_file = Path(run_netlist_file)
>         run_netlist_file = run_netlist_file.with_suffix(".asc")
>         with open(run_netlist_file, 'w') as asc_file:
>             _logger.info(f"Writing ASC file {run_netlist_file}")
>             asc_file.writelines(self._asc_file_lines)
> ```
> 
> In `tolerance_deviations.py`
> 
> ```python
>     def save_netlist(self, filename: str):
>         if self.testbench_prepared is False:
>             self.prepare_testbench()
>         self.editor.write_netlist(filename)
> ```

So I think in this circumstance, you cannot save the netlist in `.net` format.

---

By the way, I'm pretty curious if the sim toolkits can be used when operating with `.net` file. From the source code, I infer that it may work because the MonteCarlo class can accept `BaseEditor` instance as its argument when inited.

---

I'm currently writing an easy-to-use framework based on PyLTSpice package. The working flow of this framework is currently like this:

1. Open and manipulate the netlist file, and register it to a list.

2. Run the simulator on all the netlist files in the registered list (in threads).

3. Obtain all the data and step traces from all the netlists that were run previously.

4. Based on the data we get, user can define their own data process methods.

My question is, do you recommend adding an optional step that setup the sim-toolkits before or after manipulating the netlist file?

Thank you.

nunobrum, 2023-09-12:
Hi there,

You are right. The .net extension there is misleading. Certainly a result of a copy+paste followed by a poor review.
I regret the inconvinience.
I strongly advise you to switch to spicelib. PyLTSpice will be gradually scaled down and depend on the spicelib.
In spicelib, you will find a better version of PyLTspice, unfortunately, compatibility with PyLTspice is not assured, being this the reason of the move.
In regards to the analysis toolkit, it aims to be an integrated solution that implements the following steps :
1. preparing testbench
2. running simulation
3. gathering data

In spicelib examples you can see an example of Montecarlo and extreme value analysis. (step 3 not implemented)

It's still far from the final goal, but, if you want to cooperate, all help is hightly appreciated.
What do you say? Are you interested in helping.

Best Regards,
Nuno

xgzi, 2023-09-14:
@nunobrum 

Thanks for your detailed answer and for inviting me to cooperate.
However, my Python skill is only at the primary level, and currently, I'm facing the pressure of seeking jobs and graduating.

It's a pleasure for me to do something in this project. Specifically, I will report any issues and suggestions I find.

closed 2023-09-14 by xgzi