nunobrum/PyLTSpice #126: Could not open input deck for reading

gverez, 2023-10-18:
Hello,

On Windows 11 Pro 22H2, LTSpice (x64): 17.1.15, PyLTSpice 5.0.
I want to run a batch computation from a .asc file, inside a Jupyter Notebook or from a Python file directly in the prompt, same results, things seem to launch but the simulation is way too fast, and in fact nothing is done.

A .net file is created, which seems correct, with all the values created EXCEPT for one line, comparing that on another computer with Windows 10, it is missing ".tran 0 205m 200m 10u startup" before
.backanno
.end
that appear at the end of the .net file. 

The .log file reports an error:
C o u l d   n o t   o p e n   i n p u t   d e c k   f o r   r e a d i n g :   path /filename . l o g 
 
This is the script I'm using, whatever the simulation I try, I get the same problem. It's not a folder problem, PyLTSpice finds the .exe file and the .asc file, as well as the Test folder.

```
filename = "blabla"
from PyLTSpice import SimRunner, SpiceEditor
simulator = r"C:\Program Files\ADI\LTspice\LTspice.exe"
runner = SimRunner(output_folder='Test', simulator= simulator)  # Configures the simulator to use and output
netlist = SpiceEditor("Test/" + filename + ".asc")  # Open the Spice Model, and creates the .net
raw, log = runner.run_now(netlist, run_filename= filename)
```

Thanks for your help!

nunobrum, 2023-10-18:
Hello!

I've tried your code, just changing the filename and the simulator location and I didn't have the same issue.
A few questions pop into mind.

-  has the "blabla" string a path in it ? (This was a problem I had and it lead me to a bug that I just corrected)
- do you have read and write access to all the paths that you are operating ?
- do you  have a file with the name Test ?

Other than that, I can't really help you more. 

Best Regards,
Nuno

gverez, 2023-10-19:
Hello Nuno, thanks to your inputs, I managed to make things work!
So, my string named filename didn't have a path in it, it was just the name of my .asc file. Then I tried with full paths for both folders and files names, and now it works.
Thank you for maintaining this very useful library,
Best regards

closed 2023-10-19 by gverez

AnasMalas, 2023-10-31:
I'll leave this here for anyone in the future who may run into the same symptoms as me, I got the same log message and all my simulations said they were successful but they werent. Using the Nuno's list I saw that a new .net file is not being created if I called the function as ```runner.run(netlist, callback=processing_data, run_filename=run_netlist_file)```, but worked fine when I called it as ```runner.run(netlist, callback=processing_data)```.

Turns out my issue was that due to ```run_netlist_file = "file.net"```. Rewriting it as ```run_netlist_file = "file.asc"``` solved it. This might be because im using AscEditor instead of SpiceEditor (```netlist = AscEditor(sim_file_path)```)