nunobrum/PyLTSpice #127: Issues running the example

Ottega, 2023-10-23:
Hey,

I'm trying to run the [SpiceEditor, AscEditor and SimRunner.py](https://github.com/nunobrum/PyLTSpice#spiceeditor-asceditor-and-simrunnerpy) example. But I get the following erron on line 6 `LTC.create_netlist('Batch_Test.asc')`

[image]

```
PS C:\Users\VirtualPC\src\pyltspice_test> & C:/Users/VirtualPC/AppData/Local/Programs/Python/Python312/python.exe c:/Users/VirtualPC/src/pyltspice_test/main.py
Failed to create netlist
Traceback (most recent call last):
  File "c:\Users\VirtualPC\src\pyltspice_test\main.py", line 6, in <module>
    LTC.create_netlist('Batch_Test.asc')
  File "C:\Users\VirtualPC\AppData\Local\Programs\Python\Python312\Lib\site-packages\PyLTSpice\sim\sim_runner.py", line 161, in create_netlist
    return self.simulator.create_netlist(asc_file, cmd_line_switches=cmd_line_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\VirtualPC\AppData\Local\Programs\Python\Python312\Lib\site-packages\spicelib\simulators\ltspice_simulator.py", line 200, in create_netlist
    raise RuntimeError(msg)
RuntimeError: Failed to create netlist
```

I'm using:
- Windows 10
- Python 3.12.0
- LTSpice 17.1.15

I tried:
- using a empty `temp` directory
- creating an empty `Batch_Test.asc` using LTSpice
- adding some components to `Batch_Test.asc` using LTSpice
- Installing LTSpice for my user only
- Installing LTSpice for all users

Do you have any idea how to resolve this issue?

Best,
Ottega

nunobrum, 2023-10-23:
Can you precise which PyLTSpice are you using?

Ottega, 2023-10-24:
That would have been useful information...
I've updated the orginal post. I'm running the latest version 5.0

nunobrum, 2023-10-24:
Thanks. Now I'm mobile and I can check it myself. But there are two things you can try. 

From the log I see that the problem is on the generation of the netlist. 

So, You can try to generate the netlist by LTSpice and see if is well generated.

Second thing, is to rather use the examples in spicelib. 
PyLTSpice os today just a skin of spicelib where the user doesn't need to specify the simulator. 

I'll update this issue as soon as I'm able to test this myself.

Ottega, 2023-10-25:
I got it running.
It was my mistake.
I did not have the `Batch_Test.asc` in the base folder, I had it in the `./temp`... sorry for the confustion.

closed 2023-10-25 by Ottega

flyforward2016, 2024-01-23:
@Ottega I have exactly the same issue as you did. How did you resolve it? My understanding the script should create a `Batch_Test.asc` itself.