nunobrum/PyLTSpice #128: Worst Case Analysis runs too many simulations

gudnimg, 2023-10-26:
In the `WorstCaseAnalysis` class, if a component does not have any tolerance assigned via for example `wca.set_tolerance('C2', 0.05)`
then the component will be given 0% tolerance.

```
self.device_deviations = {'C2': ComponentDeviation(max_val=0.05, min_val=-0.05, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform')}
self.parameter_deviations = {}
self.default_tolerance = {
'R': ComponentDeviation(max_val=0, min_val=0.0, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform'),
'C': ComponentDeviation(max_val=0, min_val=0.0, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform'),
'L': ComponentDeviation(max_val=0, min_val=0.0, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform'),
'V': ComponentDeviation(max_val=0, min_val=0.0, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform'),
'I': ComponentDeviation(max_val=0, min_val=0.0, typ=<DeviationType.tolerance: 'tolerance'>, distribution='uniform')
}
```

This yields 4095 simulations! In this scenario I expect only 2 simulations for C2 at +5% and -5%

The problem is `prepare_testbench` increments `index` in the calculation for number of simulations to run: `self.num_runs = 2**index - 1`, even if the component has no tolerance set. In my use case, I don't set tolerance to a component I know has no effect on the circuit in order to reduce simulation time.

Would it be a reasonable solution to not increment `index` if `ComponentDeviation` class has `max_val = 0` and `min_val = 0`?

Example from here https://github.com/nunobrum/PyLTSpice/blob/master/examples/run_worst_case.py should suffice to reproduce this.

Remove these lines:

```python
wca.set_tolerance('R', 0.01)  # 1% tolerance
wca.set_tolerance('C', 0.1)  # 10% tolerance
wca.set_tolerance('V', 0.1)  # 10% tolerance. For Worst Case analysis, the distribution is irrelevant
```

⚠️ And change `wca.run()` into `wca.run(1)` somehow this helps trigger the problem.

gudnimg, 2023-10-26:
There is also an issue with `range(-1, self.num_runs, max_runs_per_sim)` in `ToleranceDeviations` class `run()`

Using the example https://github.com/nunobrum/PyLTSpice/blob/master/examples/run_worst_case.py 

And removing only these lines:

```python
wca.set_tolerance('R', 0.01)  # 1% tolerance
wca.set_tolerance('C', 0.1)  # 10% tolerance
wca.set_tolerance('V', 0.1)  # 10% tolerance. For Worst Case analysis, the distribution is irrelevant
```

But keep this code code:

```
# Some components can have a different tolerance
wca.set_tolerance('R1', 0.05)  # 5% tolerance for R1 only. This only overrides the default tolerance for R1

# Tolerances can be set for parameters as well.
wca.set_parameter_deviation('Vos', 3e-4, 5e-3)

```
This will only run one simulation! I'd expected more.

Because `range(-1, self.num_runs, max_runs_per_sim)` with `self.num_runs = 9` and `max_runs_per_sim = 512` yields only `-1`
It should produce iterable of length `self.num_runs`

closed 2023-10-27 by gudnimg

nunobrum, 2023-10-27:
Thanks for your contact. 

I understood that there are situations were the user wants to restrict the worst case to just a few components. 

I should default to 0% deviation in all components, unless user reverts it. In this case, only components with !=0 deviation should be considered. 

Best regards,
Nuno

nunobrum, 2023-11-05:
A new version has been pushed that has some more fixes introduced to the toolkit.