nunobrum/PyLTSpice #129: Issues running the example - 'gbk' codec can't decode byte 0xb5 in position 3537: illegal multibyte sequence

flyforward2016, 2024-01-23:
Hey,  I was trying to run the sim_runner_asc_example.py and it throw an error as below.  Could you shine some light on it? Thank you very much

```
\PyLTSpice\lib\site-packages\spicelib\editor\asc_editor.py", line 50, in __init__
    self.reset_netlist()
  File "C:\Anaconda3\envs\PyLTSpice\lib\site-packages\spicelib\editor\asc_editor.py", line 67, in reset_netlist
    self._asc_file_lines = asc_file.readlines()
UnicodeDecodeError: 'gbk' codec can't decode byte 0xb5 in position 3537: illegal multibyte sequence`

```

Here is the my configuration

Windows 10
Python 3.8
LTSpice 17.0.37.0
PyLTSpcie 5.2

nunobrum, 2024-02-15:
Oh!  I've missed this notification and only seeing this issue now.  I hope you can excuse me this.
I've been moving much of the code from PyLTSpice to spicelib. Keeping PyLTSpice only for backwards compatibility and for simplifying things for users that don't want to use more than one simulator, relieving them from the burden of specifying the simulator.
The example you are using should have been updated in spicelib but not in PyLTSpice. 
The method that should be called is run_analysis() instead of just run()

nunobrum, 2024-02-15:
I've just pushed a new version that fixes the problem you were facing.
The update on the PyPi repository will follow as soon as I also finish some fixes on the spicelib base repository.

closed 2024-02-15 by nunobrum