nunobrum/PyLTSpice #135: sim_runner_callback_process_example.py Encounters an Error When Opening Batch_Test.asc

mago3421, 2024-03-21:
Python version: 3.12
LTSpice version: 24

When running the example, sim_runner_callback_process_example.py, an error occurs in the below code:

`    netlist = SpiceEditor("./testfiles/Batch_Test.asc")  # Open the Spice Model, and creates the .net
`

The error reports that incorrect header characters are used:
`  File "C:\Users\mgonza17\Documents\lts_automation\lts\Lib\site-packages\spicelib\utils\detect_encoding.py", line 77, in detect_encoding
    raise EncodingDetectError(f"Expected pattern \"{expected_pattern}\" not found in file:{file_path}")
spicelib.utils.detect_encoding.EncodingDetectError: Expected pattern "^\* " not found in file:testfiles\Batch_Test.asc`

So, I created a new .ASC file in LTSpice but see the same error:

`  File "C:\Users\mgonza17\Documents\lts_automation\lts\Lib\site-packages\spicelib\utils\detect_encoding.py", line 77, in detect_encoding
    raise EncodingDetectError(f"Expected pattern \"{expected_pattern}\" not found in file:{file_path}")
spicelib.utils.detect_encoding.EncodingDetectError: Expected pattern "^\* " not found in file:..\..\test_RC.asc`

Why is this example not working?

The error occurs in spice_editor.py:
`                try:
                    self.encoding = detect_encoding(self.netlist_file, '^\* ')  # Normally the file will start with a '*'
                except EncodingDetectError as err:
                    raise err`

Does PyLTSpice not work with LTSpice 24?

nunobrum, 2024-03-21:
It's not possible to use SpiceEditor with an ASC file. 
You must use AscEditor instead. 
This was something that changed a while ago. Maybe examples need to be revised.

mago3421, 2024-03-21:
Thanks for your response. Is PyLTSpice completely compatible with version 24?

nunobrum, 2024-03-24:
Yes. I have been testing with LTspice 24 in windows and it works well. 
I haven't tested in other Platforms such like Linux and Mac OSX.

closed 2024-03-24 by nunobrum