nunobrum/PyLTSpice #136: example sim_runner_callback_example.py needs the "rich" module

viisatadi, 2024-03-22:
Given that the examples use this module, it should be an included requirement

Error message is:
```
python sim_runner_callback_example.py
Traceback (most recent call last):
  File "sim_runner_callback_example.py", line 3, in <module>
    from rich.logging import RichHandler
ModuleNotFoundError: No module named 'rich'
```

nunobrum, 2024-03-24:
I'll probably remove this feature from the example on the next version. 
I prefer not to add yet another dependency. I'll close this issue when the  next version is released.

nunobrum, 2024-04-28:
The import rich has now a try: except ImportError: clause around it. If it is not found, then it is not used.

closed 2024-04-28 by nunobrum