nunobrum/PyLTSpice #138: Issue after change SpiceEditor to ASCEditor following #135

flyforwardFast, 2024-05-02:
Following this issue https://github.com/nunobrum/PyLTSpice/issues/135, the suggestion was to use `AscEditor`.  
1. I have run into below issues for `sim_runner_callback_example.py`  and  `sim_runner_callback_process_example.py ` and `sim_stepper_example.py` and all showing the same error ` spicelib.editor.base_editor.ComponentNotFoundError: Component XU1 not found in Schematic file. `. If i change `XU1` to `U1`, it throws an error `spicelib.editor.base_editor.ComponentNotFoundError: Component U1 does not have a Value attribute`
2. `spice_editor_example.py` throws an error `AttributeError: 'AscEditor' object has no attribute 'run'`

nunobrum, 2024-05-03:
I've traced this to the fact that symbols with X prefix can be either subcircuits inside of library functions or other ASC files on a hierarchical design. PyLTspice (spicelib) presently do not support this.
I am fixing this and will release soon a new version that fixes this problem.
Thanks in advance for your patience.

nunobrum, 2024-05-05:
I have fixed these two issues. 
Hierarchical edits are possible, the referenced .asc is going to be edited and saved.
Just want to note that making possible to change spice libraries from within an ASC file manipulation is not possible. If this is needed, I suggest then to generate the spice netlist first and make all the edits on top of it.

Please let me know if you find any other problems. 

Thanks for your patience,
Nuno

closed 2024-05-05 by nunobrum

flyforwardFast, 2024-05-06:
Hi Nuno, 
Thanks much for your effort and time in the newest release. However, the issues in the examples still present and hope to get your input. 

1. Following this issue `https://github.com/nunobrum/PyLTSpice/issues/135`, We have to modify `SpiceEditor` to `AscEditor`. Please instruct if this is still needed. (all below testings were done based on this assumption)

2. All the errors in the examples are still there using the most recent release(5.3.1). (following suggestion in #1) 

-  `spice_editor_example.py` -> AttributeError: 'AscEditor' object has no attribute 'run'

-  `sim_runner_callback_process_example` and `sim_stepper_example` -> spicelib.editor.base_editor.ComponentNotFoundError: Component XU1 not found in Schematic file

- `sim_runner_example.py` -> NotImplementedError: Primitive not supported for ASC file

nunobrum, 2024-05-09:
I don't know how you have your setup configured. Maybe there is a problem with your environment and you have still some old spicelib or PyLTspice installed in your PC.

I suggest you set first the environment PYTHONPATH to the directory where you have the checked out or installed the pyLTspice.
```
C:\sandbox\PyLTSpice\examples>set PYTHONPATH=C:\sandbox\spicelib\;C:\sandbox\PyLTSpice\
```

Then when you invoke the example it should run well as it does in my PC
```
C:\sandbox\PyLTSpice\examples>py -3.8 C:\sandbox\PyLTSpice\examples\sim_runner_example.py
simulating OpAmp AD712 Voltage 5
simulating OpAmp AD712 Voltage 10
simulating OpAmp AD712 Voltage 15
simulating OpAmp AD820 Voltage 5
simulating OpAmp AD820 Voltage 10
simulating OpAmp AD820 Voltage 15
Raw file: temp\Batch_Test_1.raw, Log file: temp\Batch_Test_1.log
Raw file: temp\Batch_Test_2.raw, Log file: temp\Batch_Test_2.log
Raw file: temp\Batch_Test_3.raw, Log file: temp\Batch_Test_3.log
Raw file: temp\Batch_Test_4.raw, Log file: temp\Batch_Test_4.log
Raw file: temp\Batch_Test_5.raw, Log file: temp\Batch_Test_5.log
Raw file: temp\Batch_Test_6.raw, Log file: temp\Batch_Test_6.log
Successful/Total Simulations: 6/6
Press enter to delete created files
```