nunobrum/PyLTSpice #146: Path to default ltspice install missing

Pimss, 2025-01-31:
*Version of PyLTSpice : 5.4.0*
*Version of LTSpice : LTspiceXVII*

The path to a windows default install is not provided, resulting in a need for user to provide path to install .exe

Possible fix : 
- Add ```"C:\Program Files (x86)\LTC\LTspiceXVII\XVIIx64.exe"``` to ```_spice_exe_win_paths``` in [ltspice_simulator.py](https://github.com/nunobrum/spicelib/blob/main/spicelib/simulators/ltspice_simulator.py)
- Detect an existing LTspice.exe, XVIIx64.exe or scad3.exe. Sample code:

```
import os

def find_exe_file(file_name, start_dir='C:\\'):
    for root, dirs, files in os.walk(start_dir):
        if file_name in files:
            return os.path.join(root, file_name)
    return None

exe_name = 'LTspice.exe' 

exe_path = find_exe_file(exe_name)

if exe_path:
    print(f"Found .exe file at: {exe_path}")
else:
    print(f"{exe_name} not found.")
```

Pimss, 2025-01-31:
@nunobrum, just wanted to add that I have been using this tool for more that 3 years and I trully appreciate all the work.

nunobrum, 2025-02-08:
Sorry I have missed this in the latest spicelib release.
I think I prefer the first option. Depending on the install, having a search for the whole disk can take ages, and could not find the LTSpice version you want.
In any case, you can in the mean time solve the issue by using the following approach.

```
class MySpiceInstallation(LTspice):
    spice_exe = [r'C:\Program Files (x86)\LTC\LTspiceXVII\XVIIx64.exe']
```
or 
`myltspice = LTspice.create_from(r"C:\Program Files (x86)\LTC\LTspiceXVII\XVIIx64.exe")`

In the future, please consider posting your issues in spicelib. There is more attention to the issues there.

Pimss, 2025-02-11:
I agree on the first solution been the most effective here.
The tests I have performed with the recursive method can be rather slow.
I thought that it could be added as a tool to find a working install. 
This is however not needed and rather simple to code outside pyLTSpice or spicelib. 

I am closing this issue, as this is fixed with commit d1f38bb in spicelib

closed 2025-02-11 by Pimss