nunobrum/PyLTSpice #148: update documentation

samancyrus-eaton, 2025-05-09:
You might want to replace 
```python
netlist = SpiceEditor("Batch_Test.asc")
```
with 
```python
netlist = SpiceEditor('./testfiles/Batch_Test.net')
``` 
in [this][https://github.com/nunobrum/PyLTSpice/blob/master/doc/modules/run_simulations.rst]. I don't think `.asc` can be loaded with `SpiceEditor()`, can it?

nunobrum, 2025-05-11:
It depends. 
If you use the SpiceEditor class that is on spicelib, then no, this is **not** possible. 
But, if you use the SpiceEditor class that is on the PyLTSpice project then **yes** ou can call it directly with the .asc file.

In fact this is the difference between the two libraries, and why the existence of the  PyLTspice library is still justified.

samancyrus-eaton, 2025-05-12:
I can see 
```python
if netlist_file.suffix == ".asc":
LTspice.create_netlist(netlist_file)
netlist_file = netlist_file.with_suffix(".net")
```
In `PyLTSpice\editor\spice_editor` inside this github repo, but on my machine when I try to call `net = SpiceEditor("myfile.asc")` it simply takes me to a file which has this:
```python
def __init__(self, netlist_file: Union[str, Path], encoding='autodetect', create_blank=False):
        super().__init__()
        self.netlist_file = Path(netlist_file)
        if create_blank:
            self.encoding = 'utf-8'  # when user want to create a blank netlist file, and didn't set encoding.
        else:
            if encoding == 'autodetect':
                try:
                    self.encoding = detect_encoding(self.netlist_file, r'^\*')  # Normally the file will start with a '*'
                except EncodingDetectError as err:
                    raise err
            else:
                self.encoding = encoding
        self.reset_netlist(create_blank)
```
The problem is that when I get to the point when I'm running:
```python
self.encoding = detect_encoding(self.netlist_file, r'^\*')
```
I get this error:
```python
EncodingDetectError: Expected pattern "^\*" not found in file:myfile.asc
```

nunobrum, 2025-05-14:
I guess you are importing both flies from spicelib and from PyLTSpice.

```python
class SpiceEditor(SpiceEditorBase):

    def __init__(self, netlist_file: Union[str, Path], encoding='autodetect', create_blank=False):
        netlist_file = Path(netlist_file)
        if netlist_file.suffix == ".asc":
            LTspice.create_netlist(netlist_file)
            netlist_file = netlist_file.with_suffix(".net")
        super().__init__(netlist_file, encoding, create_blank)

    def run(self, wait_resource: bool = True, callback: Callable[[str, str], Any] = None, timeout: float = 600,
            run_filename: str = None, simulator=None):
        if simulator is None:
            simulator = LTspice
        return super().run(wait_resource, callback, timeout, run_filename, simulator)
```

nunobrum, 2025-07-28:
Closing this issue as it seem stalled with pending question to the reporter.

closed 2025-07-28 by nunobrum