nunobrum/PyLTSpice #150: Install Error - clipboard

the-moog, 2025-07-21:
This package, as installed from pip, pulls [clipboard](https://github.com/terryyin/clipboard) as a dependency.

It appears clipboard uses outdated distutils so can't install as pip tried to build a wheel.

Perhaps remove this dep or use [paperclip](https://pypi.org/project/paperclip/) instead?

nunobrum, 2025-07-28:
Hello! Thanks for reporting this.
I'll see what is the best way to sort this. 
By the way, I guess you mean pyperclip, not paperclip.
I actually moved from pyperclip to clipboard. pyperclip is a nice library, but the owner seems to no longer maintains it. I have a PR there for more than 2 years already.

the-moog, 2025-07-30:
I'm working on a solution.  Hopefully have it working soon and when it does will PR.

the-moog, 2025-08-01:
I've put a pull request against spicelib, take a look and let me know if it's any use.

nunobrum, 2025-08-10:
The 1.4.5 was just released containing the new clipboard functions, so, I'm closing this issue.
Regarding the pyltspice-build, I think README.md the needs to be further developed. 
By reading it I was not able to understand what the library does.

closed 2025-08-11 by nunobrum